chan-sccp/chan-sccp #564: 7960s don't register completely

sgofferj, 2021-02-02:
I installed asterisk and chan-sccp after a hardware failure on a new Raspberry Pi. My 7960s don't register completely. sccp show devices show their regstate as "Progress" and depending on the asterisk version, interacting with the phones leads to deadlocks of the asterisk.

**Reproduction Steps:**
unclear

**chan-sccp version:** (include revision and/or branch if relevant)
Tested with release 4.3.3 and latest git

**asterisk version:** (include revision and/or branch if relevant)
tested with 16.16, 18.1.1 and 18.2.0

**Expected behavior:**
Phones register

**Observed behavior:**
Phones don't register

**Screenshots**

(Screenshots which show reproduction steps and/or demonstrate the problem)
[url]

**Addition / Optional Information:**
- Problem started happening recently, didn't happen in an older version of Chan-SCCP: Yes
- Problem can be reliably reproduced, doesn't happen randomly: No
- Platform:
  - Raspberry Pi 3B
  - 64 Bit Ubuntu 20.04
 
Only debug info I got so far is
```
    -- SCCP: Alarm Message: Severity: Major (7), 27: Waiting for Requesting Feature Key Status From 192.168.10.203 [14/-888493888]
```
with .203 being th eIP of the asterisk.

sgofferj, 2021-02-02:
[image]

sgofferj, 2021-02-02:
7905 registers fine and works but the speeddials from the config don't show up in the phone.

sgofferj, 2021-02-02:
Recompiled now with asterisk 18.2.0 and latest git and same symptoms.Without the devstate button, the phones register fine. With the devstate button, they hang. I captured a full asterisk startup with the requested debug params:

[asterisk_start.log](https://github.com/chan-sccp/chan-sccp/files/5911296/asterisk_start.log)

About using a release from earlier this year - I actually used the 4.3.3 release from December last year for testing :).

dkgroot, 2021-02-02:
After the devstate changes (middle last year), i ran all the tests on > 7961 and 89xx. Which would explain why it was not uncovered.
The registration process getting as far as it does would indicate the devices don't like one of the messages being sent to them.
Sorry it will take me a little bit of time before i can have a look and test it out on a physical 7960. Happy to see that without the devstate it works fine.

The secondary issue where asterisk is locked up when you dial a partially registered phone will have to researched as well, and certainly needs fixing too.

sgofferj, 2021-02-03:
If you tell me how to get a debug log from a 7960, I can probably produce one.