chan-sccp/chan-sccp #565: Is video over IAX trunk possible?

Zw1d, 2021-02-04:
Hi!
This is most likely not an issue with chan-sccp. I'm just hoping to get some clarification.
I have two PBXs in two locations (A and B) connected via IAX2 trunk. Local video calls work in both site A and in site B, but they do not work across the trunk (eg. call from site A to site B):

`SCCP => IAX-Trunk => SCCP`

This is the message I managed to pick up on receiving end:

`[2021-02-04 11:54:28] NOTICE[3631][C-00000a76]: ast116.c:1706 sccp_astwrap_request: Could not match video codec. No Video`

I know that chan-sccp support h263 and h264. Is that possible that stream from SCCP is converted to some other codec that IAX trunk uses, that cannot get reverted back to be received by SCCP on the other side?

Thank you in advance and sorry for not exactly chan-sccp related question.

ojaksch, 2021-02-04:
Never tried this due to the lack of video phones. There was a parameter "videosupport=yes" in iax.conf on older versions of Asterisk but seems to be gone. Also, check the allowed/disallowed codecs in iax.conf. There are some in the [general] section as well in the friends/peers section.
To name a few possible codecs: h263, h263p, h264
Just my 2c.

Zw1d, 2021-02-04:
@ojaksch Thanks for the input. `videosupport=yes` is available in general IAX configuration and I think it works okay, but something happens to the stream that goes through the trunk so it's not accepted on the other end.

Zw1d, 2021-02-05:
No progress so far, but I can confirm that it does work over SIP trunk. However SIP trunk config is a nightmare and I hate it.

dkgroot, 2021-02-06:
@zipiju Can you run a little video test using two sip softphones over that IAX trunk. If that doesn't work either, we know it's an iax problem and chan-sccp is not involved. If it does work, then it points to a problem in the way chan-sccp and iax work together, to figure out codec support. Thanks for reporting this issue and testing it out.