chan-sccp/chan-sccp #566: No speddials for 7905

sgofferj, 2021-02-04:
I installed asterisk and chan-sccp after a hardware failure on a new Raspberry Pi. My 7905 doesn't show any speeddials although they are configured in sccp.conf. That used to work.

**Reproduction Steps:**
Setup speeddails for a 7905

**chan-sccp version:** (include revision and/or branch if relevant)
Tested with release 4.3.3 and latest git

**asterisk version:** (include revision and/or branch if relevant)
tested with 16.16, 18.1.1 and 18.2.0

**Expected behavior:**
Phone has speeddials

**Observed behavior:**
Phones doesn't have speeddials

**Screenshots**
[image]

**Addition / Optional Information:**
```
[SEP000D6508DF61](7905)
description		= Maiju
button			= line,6004
button			= speeddial,Desk,**6000,6000@internal
button			= speeddial,Sofa,**6003,6003@internal
button			= speeddial,Bedroom,**6002,6002@internal
button			= speeddial,Stefan mobile,6100,sgofferj@internal
```
- Problem started happening recently, didn't happen in an older version of Chan-SCCP: Yes
- Problem can be reliably reproduced, doesn't happen randomly: Yes
- Platform:
  - Raspberry Pi 3B
  - 64 Bit Ubuntu 20.04

dkgroot, 2021-02-05:
@sgofferj I think i know which revision caused it.
I think the problem comes from revisions: bbeed39 and 31c25a.
Could you checkout e0cb339 to confirm that. If that works i should start by reverting these two changes.
Sorry for the inconvenience !

dkgroot, 2021-02-06:
I think this might be directly related to #564. If you can confirm that everything was fin in this regard when you use revision: e0cb339 we should be able to undo those changes quickly and get everything back on track. To test it, it should be as simple as:
`git checkout e0cb339` and then `make -j2 && make install && make reload`

sgofferj, 2021-02-09:
Will do. Going to take a few days, though - lots on my list atm.