chan-sccp/chan-sccp #567: FRACK!, Failed assertion bad magic number 0x0 for object

abradfordcms, 2021-02-13:
Asterisk crashes with ERROR[24577][C-00000058]: pbx_variables.c:1084 pbx_builtin_setvar_helper: FRACK!, Failed assertion bad magic number 0x0 for object 0x7fc4d0020330 (0) 

**Reproduction Steps:**

1. End call via hang up on a Cisco 7975 on commit 16fd7da

**chan-sccp version:** (include revision and/or branch if relevant)

4.3.3 develop - 16fd7da 

**asterisk version:** (include revision and/or branch if relevant)

16.15.1

**Expected behavior:**

Not crashing :) 

**Observed behavior:**

Appears that when a 7975 hangs up a call, this is the last bit of the Asterisk CLI trace before a crash: 

    -- SEPF09E630FF5D1: is Onhook (buttonIndex: 0, callid: 0)
  == SEPF09E630FF5D1: Ending call SCCP/404-00000071 (state:CONNECTED)
    -- SEPF09E630FF5D1: Sending hangupRequest to Call SCCP/404-00000071 (state: CONNECTED)
  == MixMonitor close filestream (mixed)
[2021-02-12 15:59:11] WARNING[13049][C-00000058]: channel.c:2282 ast_channel_destructor: PBX may not have been terminated properly on 'SCCP/404-00000071'
  == End MixMonitor Recording SCCP/404-00000071
    -- QualityStats: MLQK=4.5000;MLQKav=4.4433;MLQKmn=3.9155;MLQKmx=4.5000;ICR=0.0000;CCR=0.0016;ICRmx=0.0434;CS=11;SCS=3;MLQKvr=0.95
    -- SEPF09E630FF5D1: Call Statistics:
       [
         Last Call        : CallID: 113 Packets sent: 9854 rcvd: 9834 lost: 1 jitter: 0 latency: 0
         Last Quality     : MLQK=4.5000;MLQKav=4.4433;MLQKmn=3.9155;MLQKmx=4.5000;MLQKvr=0.95|ICR=0.0000;CCR=0.0016;ICRmx=0.0434|CS=11;SCS=3
         Mean Statistics  : #Calls: 16 Packets sent: 3409 rcvd: 3395 lost: 0 jitter: 0 latency: 0
         Mean Quality     : MLQK=3.3476;MLQKav=3.3150;MLQKmn=3.1784;MLQKmx=4.5000;MLQKvr=0.95|ICR=0.0024;CCR=0.0030;ICRmx=0.0434|CS=2;SCS=0
       ]
    -- <>AGI Script agi://127.0.0.1/sangomacrm.agi completed, returning 0
[2021-02-12 15:59:11] ERROR[24577][C-00000058]: pbx_variables.c:1084 pbx_builtin_setvar_helper: FRACK!, Failed assertion bad magic number 0x0 for object 0x7fc4d0020330 (0)
[2021-02-12 15:59:11] ERROR[24577][C-00000058]:   Got 17 backtrace records
# 0: /usr/sbin/asterisk(__ast_assert_failed+0x84) [0x5c4eb5]
# 1: /usr/sbin/asterisk() [0x45ef13]
# 2: /usr/sbin/asterisk(__ao2_lock+0x64) [0x45ef79]
# 3: /usr/sbin/asterisk(pbx_builtin_setvar_helper+0x111) [0x54a367]
# 4: /usr/lib64/asterisk/modules/res_agi.so(+0x11011) [0x7fc489d5f011]
# 5: /usr/lib64/asterisk/modules/res_agi.so(+0x110d6) [0x7fc489d5f0d6]
# 6: /usr/sbin/asterisk(pbx_exec+0x11c) [0x54025a]
# 7: /usr/sbin/asterisk() [0x52bc5c]
# 8: /usr/sbin/asterisk(ast_spawn_extension+0x64) [0x52fa68]
# 9: /usr/lib64/asterisk/modules/app_stack.so(+0x5429) [0x7fc4ba1cd429]
#10: /usr/sbin/asterisk(ast_app_exec_sub+0xb9) [0x4429c1]
#11: /usr/sbin/asterisk(ast_pbx_hangup_handler_run+0x173) [0x545a4b]
#12: /usr/sbin/asterisk() [0x531a5b]
#13: /usr/sbin/asterisk() [0x531ef9]
#14: /usr/sbin/asterisk() [0x5c2226]
#15: /lib64/libpthread.so.0(+0x7ea5) [0x7fc4db861ea5]
#16: /lib64/libc.so.6(clone+0x6d) [0x7fc4da9008dd] 

**Addition / Optional Information:**
- Problem started happening recently, didn't happen in an older version of Chan-SCCP: Yes
- Problem can be reliably reproduced, doesn't happen randomly: No - seems to occur every few hours, but not consistently 
- Platform:
  - Linode cloud PBX 
  - 64-bit
  - 4GB RAM 
  - 1 core on AMD EPYC 7601 
  - Sangoma Linux (based on CentOS)

- Direct Contact information: andy.bradford.cms@gmail.com
- Log Files [trace_crash_16fd7da.txt](https://github.com/chan-sccp/chan-sccp/files/5975726/trace_crash_16fd7da.txt)
- Comments: [Other Information]

dkgroot, 2021-02-24:
@abradfordcms 
I am working on this issue. it was reported before and has been pretty hard to reproduce. We are closing in on the source of the problem though. It will take a while to figure out how to fix it though.

dkgroot, 2021-02-24:
Based on quite a bit of work and testing by @tomgilabert the most likely candidate of the cause for this issue was revision: 375ef06d91. To test this assumption i would like you to test this hypothesis using:

***Git***
_Checkout this temporary branch_
```
cd /usr/src/chan-sccp
git fetch origin
git branch issues/567 origin/issues/567
git checkout issues/567
git pull
./configure [....]
make -j4 && make install && make reload
```

_move back to develop afterwards:_
```
cd /usr/src/chan-sccp
git checkout develop
git pull
./configure [....]
make -j4 && make install && make reload
```

dkgroot, 2021-02-24:
backtrace (provided by @rofferman)
```
(gdb) bt
#0  0x000000000058b758 in stasis_topic_subscribers (topic=0x0) at stasis.c:642
#1  0x000000000058d13e in publish_msg (topic=0x0, message=0x7fa018004928, sync_sub=0x0) at stasis.c:1462
#2  0x000000000058d26a in stasis_publish (topic=0x0, message=0x7fa018004928) at stasis.c:1509
#3  0x000000000059bad6 in ast_channel_publish_blob (chan=0x7fa08c009ce0, type=0x291a998, blob=0x7fa0180018e0) at stasis_channels.c:1145
#4  0x000000000059bba2 in ast_channel_publish_varset (chan=0x7fa08c009ce0, name=0x7fa13da3e164 "AGISTATUS", value=0x7fa13da3d323 "SUCCESS") at stasis_channels.c:1171
#5  0x0000000000549ad1 in pbx_builtin_setvar_helper (chan=0x7fa08c009ce0, name=0x7fa13da3e164 "AGISTATUS", value=0x7fa13da3d323 "SUCCESS") at pbx_variables.c:1114
#6  0x00007fa13da3bf14 in agi_exec_full (chan=0x7fa08c009ce0,
    data=0x7fa089563a40 "agi://127.0.0.1/attendedtransfer-rec-restart.php,PJSIP/8124254576-00000026,/var/spool/asterisk/monitor/2021/01/07/out-8128995110-3583-20210107-113548-1610040940.121.wav", enhanced=0, dead=1) at res_agi.c:4545
#7  0x00007fa13da3bfd9 in agi_exec (chan=0x7fa08c009ce0,
    data=0x7fa089563a40 "agi://127.0.0.1/attendedtransfer-rec-restart.php,PJSIP/8124254576-00000026,/var/spool/asterisk/monitor/2021/01/07/out-8128995110-3583-20210107-113548-1610040940.121.wav") at res_agi.c:4566
#8  0x000000000053f786 in pbx_exec (c=0x7fa08c009ce0, app=0x3354810,
    data=0x7fa089563a40 "agi://127.0.0.1/attendedtransfer-rec-restart.php,PJSIP/8124254576-00000026,/var/spool/asterisk/monitor/2021/01/07/out-8128995110-3583-20210107-113548-1610040940.121.wav") at pbx_app.c:492
#9  0x000000000052b166 in pbx_extension_helper (c=0x7fa08c009ce0, con=0x0, context=0x7fa08c00a6a0 "macro-hangupcall", exten=0x7fa08c00a6f0 "s", priority=6, label=0x0,
    callerid=0x7fa0180057a0 "[0x549893]\n# 4: /usr/lib64/asterisk/modules/res_agi.so(+0x10f14) [0x7fa13da3bf14]\n# 5: /usr/lib64/asterisk/modules/res_agi.so(+0x10fd9) [0x7fa13da3bfd9]\n# 6: /usr/sbin/asterisk(pbx_exec+0x11c) [0x53f7"..., action=E_SPAWN, found=0x7fa089566c58, combined_find_spawn=1) at pbx.c:2947
#10 0x000000000052ef72 in ast_spawn_extension (c=0x7fa08c009ce0, context=0x7fa08c00a6a0 "macro-hangupcall", exten=0x7fa08c00a6f0 "s", priority=6,
    callerid=0x7fa0180057a0 "[0x549893]\n# 4: /usr/lib64/asterisk/modules/res_agi.so(+0x10f14) [0x7fa13da3bf14]\n# 5: /usr/lib64/asterisk/modules/res_agi.so(+0x10fd9) [0x7fa13da3bfd9]\n# 6: /usr/sbin/asterisk(pbx_exec+0x11c) [0x53f7"..., found=0x7fa089566c58, combined_find_spawn=1) at pbx.c:4206
#11 0x00007fa15f5842ec in _macro_exec (chan=0x7fa08c009ce0, data=0x7fa089567860 "hangupcall", exclusive=0) at app_macro.c:458
#12 0x00007fa15f5855dc in macro_exec (chan=0x7fa08c009ce0, data=0x7fa089567860 "hangupcall") at app_macro.c:628
#13 0x000000000053f786 in pbx_exec (c=0x7fa08c009ce0, app=0x343f440, data=0x7fa089567860 "hangupcall") at pbx_app.c:492
#14 0x000000000052b166 in pbx_extension_helper (c=0x7fa08c009ce0, con=0x0, context=0x7fa08c00a6a0 "macro-hangupcall", exten=0x7fa08c00a6f0 "s", priority=1, label=0x0,
    callerid=0x7fa0180057a0 "[0x549893]\n# 4: /usr/lib64/asterisk/modules/res_agi.so(+0x10f14) [0x7fa13da3bf14]\n# 5: /usr/lib64/asterisk/modules/res_agi.so(+0x10fd9) [0x7fa13da3bfd9]\n# 6: /usr/sbin/asterisk(pbx_exec+0x11c) [0x53f7"..., action=E_SPAWN, found=0x7fa08956a9e8, combined_find_spawn=1) at pbx.c:2947
#15 0x000000000052ef72 in ast_spawn_extension (c=0x7fa08c009ce0, context=0x7fa08c00a6a0 "macro-hangupcall", exten=0x7fa08c00a6f0 "s", priority=1,
    callerid=0x7fa0180057a0 "[0x549893]\n# 4: /usr/lib64/asterisk/modules/res_agi.so(+0x10f14) [0x7fa13da3bf14]\n# 5: /usr/lib64/asterisk/modules/res_agi.so(+0x10fd9) [0x7fa13da3bfd9]\n# 6: /usr/sbin/asterisk(pbx_exec+0x11c) [0x53f7"..., found=0x7fa08956a9e8, combined_find_spawn=1) at pbx.c:4206
#16 0x000000000052f118 in ast_pbx_h_exten_run (chan=0x7fa08c009ce0, context=0x7fa08c00a6a0 "macro-hangupcall") at pbx.c:4236
#17 0x0000000000530f56 in __ast_pbx_run (c=0x7fa08c009ce0, args=0x0) at pbx.c:4597
#18 0x0000000000531403 in pbx_thread (data=0x7fa08c009ce0) at pbx.c:4704
#19 0x00000000005c3441 in dummy_start (data=0x7fa02c001670) at utils.c:1249
#20 0x00007fa1b6460ea5 in start_thread () from /lib64/libpthread.so.0
#21 0x00007fa1b54ff8dd in clone () from /lib64/libc.so.6
```

dkgroot, 2021-03-14:
@pnlarsson Yes that looks like the same issue, can you try/test the branch i mentioned above ?

pnlarsson, 2021-03-17:
@dkgroot  - I now use that branch on this customers pbx and before I had one segfault a day - no segfaults since using this branch.

So it looks promising.

pnlarsson, 2021-03-23:
Yepp - no segfault since I used this branch - so it has fixed that bug.

pheinemeyer, 2021-03-25:
I believe I'm experiencing the same issue on two of my sites.

dmesg result:
asterisk[25877]: segfault at 10 ip 000000000058ca64 sp 00007f4631d06f20 error 4 in asterisk[400000+2e9000]

When this error occurs, the pbx resets on it's own within 1 minute. extensions are back up and running.  The issue happens about once a day when one user hangs up, but if other extensions were active on a call, they would be disconnected.

After running the checkout issues/567 and reloading chan_sccp, the pbx runs clean however there seems to be a memory leak.  The pbx locks up after 4 days and requires manual intervention to restart the pbx.  I reverted back to 4.3.3 develop.

Asterisk 16.15.1
FreePBX 15.0.17.24
chan_sccp: 4.3.3 develop

Let me know if you want any additional logs.

bknv, 2021-03-26:
@pheinemeyer Hello. I also have a problem with the asterisk stopping after 1 week, but this was the case in previous versions as well. I have not yet been able to find out what affects this.
Do you have in the moment problem message Error or Warning in /var/log/asterisk/full?

WARNING[2818][C-00003345] pbx.c: Failed to create new channel thread
WARNING[2818][C-00003345] chan_sip.c: Failed to start PBX :( 
ERROR[2811] tcptls.c: TCP/TLS unable to launch helper thread: Cannot allocate memory
ERROR[2755] tcptls.c: TCP/TLS unable to launch helper thread: Cannot allocate memory 
WARNING[2804] pbx.c: Failed to create new channel thread

pheinemeyer, 2021-03-26:
@bknv Hello.  I have not seen the logs you identified.

When I checkout issues/567, after 4 days, the pbx hangs.  I have no logs when this happens.  I can ssh login and issue the fwconsole restart command to restore service.

For the segfault issue, I have the following var/log/asterisk/full log:

[2021-03-25 10:03:20] VERBOSE[25877][C-00000017] pbx.c: Executing [s@crm-hangup:1] NoOp("SCCP/203-00000016", "Sending Hangup to CRM") in new stack
[2021-03-25 10:03:20] VERBOSE[25877][C-00000017] pbx.c: Executing [s@crm-hangup:2] NoOp("SCCP/203-00000016", "HANGUP CAUSE: 16") in new stack
[2021-03-25 10:03:20] VERBOSE[25877][C-00000017] pbx.c: Executing [s@crm-hangup:3] ExecIf("SCCP/203-00000016", "0?Set(__CRM_VOICEMAIL=)") in new stack
[2021-03-25 10:03:20] VERBOSE[25877][C-00000017] pbx.c: Executing [s@crm-hangup:4] NoOp("SCCP/203-00000016", "MASTER CHANNEL: 1616688060.41 = 1616688060.41") in new stack
[2021-03-25 10:03:20] VERBOSE[25877][C-00000017] pbx.c: Executing [s@crm-hangup:5] GotoIf("SCCP/203-00000016", "0?return") in new stack
[2021-03-25 10:03:20] VERBOSE[25877][C-00000017] pbx.c: Executing [s@crm-hangup:6] Set("SCCP/203-00000016", "__CRM_HANGUP=1") in new stack
[2021-03-25 10:03:20] VERBOSE[25877][C-00000017] pbx.c: Executing [s@crm-hangup:7] AGI("SCCP/203-00000016", "agi://127.0.0.1/sangomacrm.agi") in new stack
[2021-03-25 10:03:20] WARNING[3564][C-00000017] channel.c: PBX may not have been terminated properly on 'SCCP/203-00000016'
[2021-03-25 10:03:21] VERBOSE[25877][C-00000017] res_agi.c: <>AGI Script agi://127.0.0.1/sangomacrm.agi completed, returning 0
[2021-03-25 10:03:21] ERROR[25877][C-00000017] pbx_variables.c: FRACK!, Failed assertion bad magic number 0x0 for object 0x7f46a815d280 (0)
[2021-03-25 10:03:21] ERROR[25877][C-00000017] : Got 17 backtrace records
# 0: /usr/sbin/asterisk(__ast_assert_failed+0x84) [0x5c4eb5]
# 1: /usr/sbin/asterisk() [0x45ef13]
# 2: /usr/sbin/asterisk(__ao2_lock+0x64) [0x45ef79]
# 3: /usr/sbin/asterisk(pbx_builtin_setvar_helper+0x111) [0x54a367]
# 4: /usr/lib64/asterisk/modules/res_agi.so(+0x11011) [0x7f4643520011]
# 5: /usr/lib64/asterisk/modules/res_agi.so(+0x110d6) [0x7f46435200d6]
# 6: /usr/sbin/asterisk(pbx_exec+0x11c) [0x54025a]
# 7: /usr/sbin/asterisk() [0x52bc5c]
# 8: /usr/sbin/asterisk(ast_spawn_extension+0x64) [0x52fa68]
# 9: /usr/lib64/asterisk/modules/app_stack.so(+0x5429) [0x7f466fbec429]
#10: /usr/sbin/asterisk(ast_app_exec_sub+0xb9) [0x4429c1]
#11: /usr/sbin/asterisk(ast_pbx_hangup_handler_run+0x173) [0x545a4b]
#12: /usr/sbin/asterisk() [0x531a5b]
#13: /usr/sbin/asterisk() [0x531ef9]
#14: /usr/sbin/asterisk() [0x5c2226]
#15: /lib64/libpthread.so.0(+0x7ea5) [0x7f46b1296ea5]
#16: /lib64/libc.so.6(clone+0x6d) [0x7f46b03358dd]

I am using Cisco 7975G phones with 7916 addon module.

bknv, 2021-03-26:
@pheinemeyer What happens in / var / log / asterisk / full when the server stops? Take a look at the previous files.
In order not to make a big discussion, I propose to contact me by mail bukan_pss@mail.ru or telegrams ID: nakuB

dkgroot, 2021-03-28:
Thanks guys for testing and reporting back. I have been abroad for three weeks and had little time to look/check back. From what i am reading above we had some mixed results but non that made the issue worse. @pnlarsson @pheinemeyer  Reported improvement. The memory leak was expected to be honest, because the original commit was created to combat that. I rather have a memoryleak then a crash. The memory leak is caused because some  (small number) of the hungup channels are never ever released.

I guess step one will have to be to commit this branch (ie: revert the original commit) and get us back in a stable condition and then try to figure out the actual cause of the memory leak and make some progress that way.

dkgroot, 2021-05-13:
Sorry it took a little longer to merge this branch. It has been pushed now though

closed 2021-05-13 by dkgroot