chan-sccp/chan-sccp #568: Cisco 7912 Connection Help

Ahh6483, 2021-02-19:
Hello there! I've got all my phones connection to the chan-sccp server working amazingly. I am having some trouble configuring the Cisco 7912, could someone help me out?
I'm struggling flashing the sccp firmware to the phone.

dkgroot, 2021-02-24:
@Ahh6483 I have no direct experience with setting up a 7912 phone myself. Can you tell me how far you have gotten ?
Do you have the `CP7912080004SCCP080108A.sbin` firmware file ? Is the phone still booting ? Is it requesting ip-address from DHCP ? Is it trying to retrieve some firmware files or non at all via tftp ?

Also do re-ask the question on our [gitter chat](https://gitter.im/chan-sccp/chan-sccp) someone there might have one of these phone and gotten it running.

Ahh6483, 2021-02-27:
Thank you for your reply!
Sure, I first had the 7912 running SIP, before I found the chan-sccp. I am unable to get it into the mode where it will retrieve the firmware files. I've tried to reset the phone many times, it gets assigned an IP address from the DHCP server, but it is not looking for the files. In the TFTP root is CP7912080004SCCP080108A.sbin, XMLDefault.cnf.xml, gk0014f28dd38a, and the gkdefault.cfg. 

I also have a 8961, I see its in the Advance Server Configuration however I cannot find much information on it. Is there a way to configure the 8961 with the chan-sccp?

dkgroot, 2021-02-27:
@Ahh6483
Ok so your XMLDefault.cnf.xml should contain the firmware reference from above, so the phone knows that to load. 
the gkdefault.txt file would have to contain something like:
```
#txt
upgradecode:3,0x601,0x0400,0x0100,192.168.10.250,0x060817A,0x170409a,CP7912080004SCCP080108A.sbin
```
reference:
```
upgradecode:3,phonetype,0x0400,0x0100,tftp_server_ip,69,image_id,image_file_name
- phonetype: is 0x401 for the Cisco Unified IP Phone 7902G, 0x501 for the Cisco Unified IP Phone 7905G, or 0x601 for the Cisco Unified IP Phone 7912G.
-  image_id is a unique 32-bit integer value. This value must be different for each upgrade. For this value, it is convenient to use 0x followed by the build date on the image file name. (For example, if the image file name is
```

And then you need to use cfgfmt.exe to convert this gkdefault.txt file to the actual gkdefault.cfg file. The 7912 phone series is a little different in this compared to other (newer) cisco models. Getting the gkdefault.txt file correct and converted to cfg is the crucial part. You can check in the tftp server log if the file is actually downloading these files.

dkgroot, 2021-02-27:
Would you mind writing a little wiki page about the 7912 process, once you get it up and running ?

evazzoler, 2021-03-09:
It seems that 7912 don't like the config ti be repeated, specially the one with the callmanager IP. It reboots in a loop... Just remove it from SEP<MAC>.XML...
And about gkdefault.cfg it must contain upgrade info and connect mode only. Search chan-sccp wiki for 7912 for the correct config.
You will reset the phone (search the **2 exact procedure) in order to let it read the gkdefault.cfg and it will read it if it won't find any other config file (please rename them or temporary move them). 
Good luck!

dkgroot, 2021-05-16:
@Ahh6483 Did @evazzoler comments help ?

Ahh6483, 2021-05-16:
Yes they did! After a little bit of messing around with TFTPd I was able to get it successfully connected. 

Thank you all!

dkgroot, 2021-05-24:
@Ahh6483 Would it be worth a little hardware 7912 article on our wiki ? Or was it just a case of RTFM ?

samam9911, 2022-01-10:
I am in same situation where the phone is going into loop mostly stuck at configuring IP. DHCP server shows and IP assigned and the tftp server shows: Jan 10 10:17:10 freepbx.sangoma.local in.tftpd[25712]: RRQ from 192.168.20.129 filename SEP001646CB4791.cnf.xml
but still unable to connect to the server. Would you please elaborate on what changes you made to tftpd to get it to work? Thanks!