chan-sccp/chan-sccp #569: Black screen on video call, but only SIP -> SCCP...

evazzoler, 2021-03-09:
Hi,
I have a Grandstream GXV3240 as SIP and a Cisco 8945 as SCCP.
All is working good except video.
When I videocall the SIP phone from the SCCP, all is good and normal (but the aspect ratio on the SCCP)
When I videocall the SCCP phone from the SIP phone, I get (not always) one frame then the screen comes black (the call remains in video mode).
The SIP phone writes ”poor network" on the display. Maybe a issue with the speed/framerate renegotiation...
I really don't know from where to start analyzing the problem.

evazzoler, 2021-03-11:
I activated the sccp rtp debug and this is the cli log:

`  -- SEPAABBCCDDEEFF: Got Start MultiMedia Transmission ACK. Status:'Media Status: OK' (0), Remote RTP/UDP:'192.168.1.37:21560', Type:DirectRTP, PassThruPartyId:4294967260, CallID:35/CallID1:35
    -- SEPAABBCCDDEEFF: Multi Media Transmission Started (State: CONNECTED[30])
[2021-03-11 19:08:15] WARNING[2927][C-0000002f]: res_rtp_asterisk.c:7506 ast_rtp_read: RTP Read too short
    -- Channel SCCP/27-00000023 left 'simple_bridge' basic-bridge <7d85af4b-c58f-42fd-89d9-72c48c1eaad0>
  == SCCP/27-00000023: (sccp_channel_closeAllMediaTransmitAndReceive) Stop All Media Reception and Transmission on device SEPAABBCCDDEEFF
    -- SCCP/27-00000023: Close receivechannel on device SEPAABBCCDDEEFF (KeepPortOpen: NO)
    -- SCCP/27-00000023: Stop mediatransmission on device SEPAABBCCDDEEFF (KeepPortOpen: NO)
    -- SCCP/27-00000023: Close multimedia receive channel on device SEPAABBCCDDEEFF (KeepPortOpen: NO)
    -- SCCP/27-00000023: Stop multimediatransmission on device SEPAABBCCDDEEFF (KeepPortOpen: NO)
       > SEPAABBCCDDEEFF: Stopping PBX audio rtp transmission on channel SCCP/27-00000023
       > SEPAABBCCDDEEFF: Stopping PBX video rtp transmission on channel SCCP/27-00000023
  == SCCP/27-00000023: (sccp_channel_closeAllMediaTransmitAndReceive) Stop All Media Reception and Transmission on device SEPAABBCCDDEEFF
    -- SCCP/27-00000023: (setEarlyRTP) OFF
  == Extension Changed 27[ext-local] new state Idle for Notify User 60 
    -- Channel SIP/60-0000002c left 'simple_bridge' basic-bridge <7d85af4b-c58f-42fd-89d9-72c48c1eaad0>
  == Spawn extension (macro-dial-one, s, 55) exited non-zero on 'SIP/60-0000002c' in macro 'dial-one'
  == Spawn extension (macro-exten-vm, s, 26) exited non-zero on 'SIP/60-0000002c' in macro 'exten-vm'`
  
  Extension 60 is the grandstream, extension 27 the cisco.
 Any idea about what more debug may activate?

evazzoler, 2021-04-05:
It seems to be something linked to the screen definition of I use a very low resolution, video will start. The strange think is both resolution and ratio will be different depending on the side the call is made.
[OT] Do you think video preview is possible on a Cisco 8945 phone? I'm asking for using it in a door system. If that's not possible, maybe 8945 is not the right choice for me...

dkgroot, 2021-07-11:
@evazzoler 
At the moment we do not have much control about the resolution in use.
I don't think preview (video transmission during early rtp) is possible, i would not know how to achieve that.