chan-sccp/chan-sccp #571: Segmentation fault -

TerAnYu, 2021-06-06:
**Reproduction Steps:**

```
cd /usr/src/chan-sccp
git checkout develop
git pull
./configure
make -j4 && make install && make reload
```

**chan-sccp version:** 4.3.4_develop_r278be366

**asterisk version:** Asterisk 16.2.1~dfsg-1+deb10u2

**Expected behavior:**

```
asterisk -cvvvvvvddddd
core reload
```
```
      reload
    -- Reloading module 'extconfig' (Configuration)
    -- Reloading module 'logger' (Logger)
 Asterisk Queue Logger restarted
    -- Reloading module 'res_odbc.so' (ODBC resource)
[Jun  6 20:27:55] NOTICE[14059]: res_odbc.c:596 load_odbc_config: Registered ODBC class 'asterisk_cdr' dsn->[asterisk_cdr_base]
[Jun  6 20:27:55] NOTICE[14059]: res_odbc.c:596 load_odbc_config: Registered ODBC class 'addressbook' dsn->[addressbook_base]
    -- Reloading module 'res_config_curl.so' (Realtime Curl configuration)
    -- Reloading module 'res_config_sqlite3.so' (SQLite 3 realtime config engine)
    -- Reloading module 'res_statsd.so' (Statsd client support)
    -- Reloading module 'cdr' (CDR Engine)
[Jun  6 20:27:55] NOTICE[14059]: cdr.c:4485 cdr_toggle_runtime_options: CDR simple logging enabled.
    -- Reloading module 'cel' (CEL Engine)
    -- CEL logging disabled.
    -- Reloading module 'dnsmgr' (DNS Manager)
    -- Reloading module 'dsp' (DSP)
    -- Reloading module 'enum' (ENUM Support)
    -- Reloading module 'features' (Call Features)
    -- Reloading module 'http' (Built-in HTTP Server)
    -- Reloading module 'indications' (Indication Tone Handling)
    -- Reloading module 'acl' (Named ACL system)
    -- Reloading module 'manager' (Asterisk Manager Interface)
    -- Reloading module 'plc' (PLC)
    -- Reloading module 'udptl' (UDPTL)
    -- Reloading module 'res_pjproject.so' (PJPROJECT Log and Utility Support)
    -- Reloading module 'res_pjsip.so' (Basic SIP resource)
[Jun  6 20:27:55] NOTICE[14091]: sorcery.c:1266 sorcery_object_load: Type 'system' is not reloadable, maintaining previous values
    -- Reloading module 'res_pjsip_authenticator_digest.so' (PJSIP authentication resource)
    -- Reloading module 'res_resolver_unbound.so' (Unbound DNS Resolver Support)
    -- Reloading module 'res_pjsip_endpoint_identifier_ip.so' (PJSIP IP endpoint identifier)
    -- Reloading module 'res_musiconhold.so' (Music On Hold Resource)
    -- Reloading module 'res_smdi.so' (Simplified Message Desk Interface (SMDI) Resource)
    -- Reloading module 'res_crypto.so' (Cryptographic Digital Signatures)
    -- Reloading module 'res_stun_monitor.so' (STUN Network Monitor)
    -- Reloading module 'res_xmpp.so' (Asterisk XMPP Interface)
    -- Reloading module 'res_rtp_asterisk.so' (Asterisk RTP Stack)
    -- Reloading module 'res_pjsip_outbound_publish.so' (PJSIP Outbound Publish Support)
    -- Reloading module 'res_pjsip_publish_asterisk.so' (PJSIP Asterisk Event PUBLISH Support)
    -- Reloading module 'res_pjsip_mwi.so' (PJSIP MWI resource)
    -- Reloading module 'chan_iax2.so' (Inter Asterisk eXchange (Ver 2))
    -- Reloading module 'chan_dahdi.so' (DAHDI Telephony w/PRI & SS7 & MFC/R2)
    -- Reloading module 'chan_motif.so' (Motif Jingle Channel Driver)
    -- Reloading module 'chan_sccp.so' ()
Segmentation fault
```

**Addition / Optional Information:**
- Problem started happening recently, didn't happen in an older version of Chan-SCCP: [Yes/No]
- Problem can be reliably reproduced, doesn't happen randomly: [Yes]
- Platform:
  - NanoPi NEO-LTS (https://www.friendlyarm.com/index.php?route=product/product&product_id=132)
  - Allwinner H3, Quad-core Cortex-A7 Up to 1.2GHz
  - Armbian 21.05.2 Buster

Any attempt to reload something will reload the service.
Also, sccp does not work any commands.

dkgroot, 2021-07-09:
@TerAnYu
Can you do the following steps:
- edit /etc/asterisk/asterisk.conf and set `dumpcore=yes`
- stop the asterisk service
- start the asterisk server manually from the commandline using `cd /tmp; asterisk -f`
- when the segmentation fault happens a core file should be produced in the /tmp folder
- start the debugger using `gdb asterisk core`, wait for it to start up and enter `bt full` when you get to the prompt.

Sorry it took me a while to get to your issue report.

TerAnYu, 2021-07-16:
@dkgroot , I apologize that I answered for a long time.
```
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/arm-linux-gnueabihf/libthread_db.so.1".
Core was generated by `asterisk -f'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0xb6811658 in pthread_rwlock_wrlock () from /lib/arm-linux-gnueabihf/libpthread.so.0
[Current thread is 1 (Thread 0xad8fe3a0 (LWP 30905))]
(gdb) bt full
#0  0xb6811658 in pthread_rwlock_wrlock () from /lib/arm-linux-gnueabihf/libpthread.so.0
No symbol table info available.
#1  0xb3949546 in sccp_reload () at chan_sccp.c:362
        readingtype = SCCP_CONFIG_READINITIAL
        returnval = 0
        __PRETTY_FUNCTION__ = "sccp_reload"
        cfg = <optimized out>
#2  0xb39db444 in module_reload () at ast116.c:3981
No locals.
#3  0x00533bd6 in ast_module_reload ()
No symbol table info available.
#4  0x005052e6 in ?? ()
No symbol table info available.
Backtrace stopped: previous frame identical to this frame (corrupt stack?)
```

dkgroot, 2021-07-16:
@TerAnYu 
- You seem to be missing some symbol table information. Can you check if you have asterisk-dbg or asterisk-debuginfo packages available on your platform ?
- The line that is clear (#1) does not seem to be capable of producing a segfault. You need to make sure that the compiled and installed version of asterisk and chan-sccp is not updated before the gdb backtrace is produced, the actual sources have to be available for an informative backtrace to be produced. Once you have the backtrace you can freely update linux / asterisk and chan-sccp.

TerAnYu, 2021-07-16:
@dkgroot, sorry, in Armbian 21.05.6 this ``asterisk-dbg`` or ``asterisk-debuginfo`` is missing :(
Yes, I am online now and can correspond to this address.

dkgroot, 2021-07-16:
Issue resolved:
chan_skinny.so was still loaded, causing chan_sccp.so to choke.

closed 2021-07-16 by dkgroot