chan-sccp/chan-sccp #572: sccp manager install error

vaDrn, 2021-06-10:
Error during installating of sccp_manager on FreePBX 
[image]

how to replicate:
1. download iso from https://www.freepbx.org/ and install as usual
2. run `yum install git asterisk16-devel autoconf automake gcc`
3. run `yum install tftp tftp-server* xinetd*`
4. download and install sccp-chan:
`cd /usr/src`
`git clone https://github.com/chan-sccp/chan-sccp chan-sccp`
`cd chan-sccp`
`./configure --enable-conference --enable-advanced-functions --enable-distributed-devicestate --enable-video`
`make -j2`
`make instal`
5. Setting the asterisk modules to be loaded through FreePBX gui
6. reboot
7. Open port 2000 following this instructions: `https://github.com/chan-sccp/chan-sccp/wiki/FreePBX_Firewall`
8. Creating mysql DB from source using `mysql -p asterisk < /usr/src/chan-sccp/conf/mysql-v5_enum.sql`
9. install `sccp manager` through FreePBX GUI:
Go to Admin -> Module Admin
Click Upload Modules
Enter: `https://github.com/chan-sccp/sccp_manager/archive/refs/heads/develop.zip`
Click Download From Web.
Click Manage Local Modules
Find and click SCCP Manager. Check Install. Click Process button
Confirm installation
**will give error shown on screenshot:**
[image]

If i try to remove `develop` version of the module and install `Stable` from this link: `https://github.com/chan-sccp/sccp_manager/archive/refs/tags/Legacy-v2.tar.gz` module installation will end successfully but Realtime DB won't be configured and corresponding files won't be crated:
[image]

asterisk version: **16.17.0**
core_sccp: **4.3.4**

steve-lad, 2021-06-11:
@vaDrn Thanks for creating this issue. Unfortunately, it should be in chan-sccp/sccp_manager. Please can you move it there, if @dkgroot has not already done.

steve-lad, 2021-06-11:
Thanks for very clear reporting. Issue was easily identified with this.

Issue is closed and fix merged into Develop.

vaDrn, 2021-06-11:
Got same error. Sccp manager download link I used:

https://github.com/chan-sccp/sccp_manager/archive/refs/heads/develop.zip

[image]

vaDrn, 2021-06-11:
still got same error. I'll close issue here, and post on https://github.com/chan-sccp/sccp_manager/issues/12

closed 2021-06-11 by vaDrn