chan-sccp/chan-sccp #574: Cyrillic symbols support issue

vaDrn, 2021-06-26:
The phone displays gibberish instead of Cyrillic characters.
Was tested on phone cisco 7960

steve-lad, 2021-06-26:
@vaDrn Before testing, forget the post and redownload sccp-manager, and build again

This issue has now been fixed in chan-sccp and should be resolved

Thanks for reporting this issue and please close this issue if it is effective

vaDrn, 2021-06-26:
@steve-lad @dkgroot 
Getting error during last update install of `chan-sccp`.

What I did:
1. I updated `sccp manager` via FreePBX GUI
2. Installed libiconv:
```
cd /usr/local/src
wget https://ftp.gnu.org/pub/gnu/libiconv/libiconv-1.16.tar.gz
tar -zxvf libiconv-1.16.tar.gz
cd libiconv-1.16
./configure --prefix=/usr/local
make
make install
```

3. Updated `chan-sccp`
```
cd /usr/src/chan-sccp
make clean
git pull
./configure --enable-conference --enable-advanced-functions --enable-distributed-devicestate --enable-video
make -j2
make install
```

results in error:
```
sccp_config.c:228:49: error: ‘struct sccp_global_vars’ has no member named ‘iconvcodepage’
 #define offsize(TYPE, MEMBER) sizeof(((TYPE *)0)->MEMBER)
                                                 ^
sccp_config.c:229:31: note: in expansion of macro ‘offsize’
 #define G_OBJ_REF(x)          offsize(struct sccp_global_vars, x), offsetof(struct sccp_global_vars, x)
                               ^
sccp_config_entries.hh:146:22: note: in expansion of macro ‘G_OBJ_REF’
  {"phonecodepage",   G_OBJ_REF(iconvcodepage),   TYPE_STRINGPTR,         SCCP_CONFIG_FLAG_NONE,      SCCP_CONFIG_NOUPDATENEEDED,  "ISO8859-1",   "Translation codepage for old phones, used to convert labels/buttonnames/descriptions from UTF-8 to this codepage, on pre-java phones. DEFAULT:ISO8859-1\n"},
                      ^
compilation terminated due to -Wfatal-errors.
make[3]: *** [libsccp_la-sccp_config.lo] Error 1
make[3]: *** Waiting for unfinished jobs....
make[3]: Leaving directory `/usr/src/chan-sccp/src'
make[2]: *** [all-recursive] Error 1
make[2]: Leaving directory `/usr/src/chan-sccp/src'
make[1]: *** [all] Error 2
make[1]: Leaving directory `/usr/src/chan-sccp/src'
make: *** [all-recursive] Error 1
```

I also tried to install `libiconv` into `/usr` like this: 
```
cd /usr/src
wget https://ftp.gnu.org/pub/gnu/libiconv/libiconv-1.16.tar.gz
tar -zxvf libiconv-1.16.tar.gz
cd libiconv-1.16
./configure --prefix=/usr
make
make install
```

as result - same error during `chan-sccp` build.

I'll try to roll back to previous `chan-sccp` update and try change `configure` file manually as @steve-lad suggested before https://github.com/chan-sccp/chan-sccp/issues/574#issuecomment-868962653

vaDrn, 2021-06-26:
I found info that `iconv` is included into `glibc-common` package for centos7
this package is preinstalled on Sangoma FreePBX
[image]

do I still require to install `libiconv` manually?

vaDrn, 2021-06-27:
I'm using Sangoma FreePBX rn, and command `yum provides */libiconv.so.2` output shows that there are no package that provides this library. And searching for CentOs forums I found info like this, quote:
```
https://forums.centos.org/viewtopic.php?t=62738
...what it says is "not found" which means that it's a file that no package that we ship provides. In turn that means you have installed something that is not designed to run on CentOS so the next question is what is asking for this file and why are you not running a CentOS version of it.
```

do I really need this library?
Cuz I have 2 year old install of Sangoma FreePBX without any 3rd party `libiconv` installs and Cyrillic symbols work just fine there.

vaDrn, 2021-06-27:
I updated `sccp-chan` without installing `libiconv` on sangoma FreePBX. Cyrillic symbols seems to be working without issues.
Thank you for the help

closed 2021-06-27 by vaDrn