chan-sccp/chan-sccp #576: Unable to load module chan_sccp on Asterisk 18.5

stergios1, 2021-07-10:
Unable to load module chan_sccp on Asterisk 18.5

**Reproduction Steps:**

1. asterisk -rvvvvvv
2. raspi4-pbx*CLI> module load chan_sccp
Unable to load module chan_sccp
Command 'module load chan_sccp' failed.
[2021-07-11 00:24:08] ERROR[9501]: loader.c:283 module_load_error: Error loading module 'chan_sccp': /usr/lib/asterisk/modules/chan_sccp.so: undefined symbol: stasis_subscribe

**chan-sccp version:** 4.3.4

**asterisk version:** Asterisk 18.5.0

**Expected behavior:**

Normally the module should be loaded

**Observed behavior:**

module does not load hence no SCCP functionality from the system

**Screenshots**

n/a

**Addition / Optional Information:**
- Problem started happening recently, didn't happen in an older version of Asterisk 16.19.0: [Yes]
- Problem can be reliably reproduced, doesn't happen randomly: [Yes]
- Platform:
  - [armv7l]
  - [32 Bit]
  - [Big/Small Endian]
  - [Linux]
  - [Rasbian 5.10.17-v7l+ #1421]
- Direct Contact information: [Stergios]
- Log Files : n/a
- Comments: [Other Information]
I cloned the git repository
build:
./configure --enable-debug --enable-conference --disable-optimization
make -j2 & install & reload

dkgroot, 2021-07-11:
@stergios1 

`stasis_subscribe` should be available in both asterisk-16 and asterisk-18 as far as i know.

Did you compile asterisk-18 manually or downloaded it a a prepackaged binary ? 
- It being raspbian i assume you manually compiled asterisk (correct?).
  - Did you run `make install-headers` during the asterisk install process. This is a required step which might be missing from some of the instruction wiki's out there.
- Did you have asterisk-16 installed beforehand ?
  -  Did you remove it's header files before compiling asterisk-18 ?
- what did the chan-sccp ./configure process report as the asterisk version detected
- Can you attach the chan-sccp config.log file.

stergios1, 2021-07-11:
> @stergios1
>
> `stasis_subscribe` should be available in both asterisk-16 and asterisk-18 as far as i know.
>
> Did you compile asterisk-18 manually or downloaded it a a prepackaged binary ?
>
>     * It [...]

Thank you Diederik! 

Yes, I did compile the asterisk 18.5 manually.
Yes, It was an asterisk-16 installation and I performed an upgrade to 18.5.

This, finally, did the trick:  **make install-headers** (btw: the mwi.h file was missing)
Now the module is loaded and phones are back to normal.

closed 2021-07-11 by stergios1

dkgroot, 2021-07-11:
@stergios1 Thanks for the update...Note when you do an upgrade like this and you still have  the old sources available before you start, then run `make uninstall` before you remove for the old sources. Make sure the /usr/include/asterisk directory is empty and then start the upgrade process. You do not want to end up with mixed headers files.
Congrats on getting it to work.
Congrats on making it work.