chan-sccp/chan-sccp #577: Module Load Failed

cww2697, 2021-07-30:
When loading the module in asterisk 16 the following error occurs:

loader.c:170 module_load_error: Error loading module 'chan_sccp.so': /usr/lib64/asterisk/modules/chan_sccp.so: undefined symbol: ast_manager_check_enabled

**Reproduction Steps:**

1. Install using v4.3.4
2. attempt to load module using "load module chan_sccp.so"

**chan-sccp version:** 4.3.4

**asterisk version:** 16.3.0

**Expected behavior:**

Module loads

**Observed behavior:**

Module failed to load with the above error message

cww2697, 2021-07-30:
Issue resolved after restart of server

closed 2021-07-30 by cww2697

dkgroot, 2021-07-30:
@cww2697 Logically chan_skinny.so was still loaded when you attempted to load chan_sccp.so. They cannot both run at the same time and chan_skinny.so does not check for chan_sccp.so when loading. Thanks for closing the issue on your own. Good luck.