chan-sccp/chan-sccp #578: Not enough calls to transfer

jeremyy294, 2021-08-07:
Receiving "Not enough calls to transfer" when trying to use "DirTrfr" and not seeing "Transfer" when engaged in a call / hold.

Using latest version from git

Asterisk 16

Appreciate any advice

dkgroot, 2021-08-14:
@jeremyy294 
Is tranfer enabled for this phone ? `sccp show device XXXXX` should have `Can Transfer : on`. What is the `Softkeyset                                   ` set to for this phone ?
If you have an active call from 1234 to 2345 on SEPxxxxxx, which you have answered, you should see the 'Transfer' button on the phone as one of the softkeys. If you have changed the softkeyset then this button might not be available in the set and state you are in. If you are using the `default` softkeyset it should be there.
Once you press `Transfer` you get a new call + dialtone and you have to enter the destination. You can not speak to the second person and when finished you can press `Transfer` again to complete the transfer.

jeremyy294, 2021-08-14:
@dkgroot 

Strangely, transfer is enabled and using default softkey set, but still cannot see Transfer when engaged in a call.

Can Transfer : on
Softkey support : yes
Softkeyset : default => default (0x33b05f0)

dkgroot, 2021-08-14:
Ooh that is strange.
Can you set `sccp debug core, action, device, line, channel, pbx, softkey, indicate` in the CLI  and perform one of these calls ? Would like to see what is going on.

closed 2023-03-10 by jeremyy294