chan-sccp/chan-sccp #579: Group Pickup does not always pick up the call

bknv, 2021-08-19:
I found strange behavior in work softkey group pickup. 
I cannot pick up the call time to time. Link on video https://youtu.be/MDFFB1GJqAI 
All calls maked from 2327 (SIP) to 2311 (SIP) and intercepted from 5578 (SCCP).

Problem with interception SCCP to SCCP i does not see, it only SIP to SCCP

I enabled  "sccp debug callinfo channel device" 
[sucessful.txt](https://github.com/chan-sccp/chan-sccp/files/7012890/sucessful.txt)
[problem.txt](https://github.com/chan-sccp/chan-sccp/files/7012891/problem.txt)

Sccp version 278be36
Asterisk 13.36.0

dkgroot, 2022-01-29:
@bknv
Sorry it took me a while to get to this ticket, new job has been quite demanding. From the problem.txt it looks like the call is picked up, and the previous partner is hungup on. But no new mediapath can be established between the parties. I will be back in NL in a couple of weeks and should be able to re-test/check this. So the issue only happens with:
- SIP(calling)-> SCCP(ringing) <- SCCP(gpickup).. Stealing a call from an SCCP phone and connecting to SIP phone
- SCCP(calling)-> SIP(ringing) <- SCCP(gpickup).. Stealing a call from an SIP phone and connecting to SCCP phone
- SCCP(calling)-> SCCP(ringing) <- SCCP(gpickup).. Stealing a call from an SCCP phone and connecting to SCCP phone
Which of this three cases would be problematic ?

bknv, 2022-01-31:
@dkgroot Hello. 
Only SCCP(calling)-> SIP(ringing) <- SCCP(gpickup).. Stealing a call from an SIP phone and connecting to SCCP phone