chan-sccp/chan-sccp #582: Call Transfer Issue

cagriaksu, 2021-09-29:
Hello,

When I try to transfer a call by pressing the transfer soft button "Key is Not Active" is displayed on the screen.

I tried re-compiling from source with --enabled-advaned-functions, but to avail, still getting the same result on both a 7962 and 7975.

Terrabytee, 2021-10-02:
Sorry I have nothing to add OP but I want to confirm the same is happening with me as well.

slipdip123, 2021-11-25:
Exact same issue here with 7962G

waffle-stomper, 2022-01-02:
I'm having the same issue on a 7962.

For me at least, the issue seems to be the line. 

I go to SCCP Connectivity > Phones Manager > [Edit Extension] > Advanced, set 'Call Transfer' to 'Yes' and click 'Save'.

When the page reloads, 'Yes' is no longer selected.

This is confirmed by `sccp show line 201`:

```
--- SCCP channel driver line settings ------------------------------------------------------------------------------------
Name                                           : 201
<snip>
Can Transfer                                   : off
```