chan-sccp/chan-sccp #583: Unable to register first SCCP phone

astifr, 2021-10-19:
Hi all, I'm testing chan_sccp with a fresh install FreePBX. I follow the guide from https://github.com/chan-sccp/chan-sccp/wiki/FreePBX and I think that I have my system properly configured.
I have one Cisco phone 7965 configured under SCCP Connectivity -> Phone Manager -> SCCP Phone
But I think that i'm missing something because I can see an error under fail2ban log,
[2021-10-19 11:41:50] NOTICE[20147] sccp_actions.c: SEP2C0BE90440B0: Rejecting device: Device Unknown

I noted that in the file sccp.conf I don't have any entry for the device. I don't if it's correct or not.

I'm using latest version of chan_sccp (downloaded yesterday) and FreePBX 15.0.17.55 with asterisk 16.

Thank you in advence.

waffle-stomper, 2021-11-25:
I'm having the same issue with a Cisco 7962G running the cmterm-7942_7962-sccp.9-4-2SR3-1 firmware on Asterisk 16.20.0/FreePBX 15.0.17.34/chan-sccp 4.3.4 develop - b0b3cdb.

/var/log/asterisk/full shows the following:

```
[2021-11-25 22:54:14] VERBOSE[2555] :     -- SCCP: Accepted Client Connection from [phone_ip]:49152
[2021-11-25 22:54:14] VERBOSE[2555] :     -- SCCP: Connected on server via [freepbx_ip]:2000
[2021-11-25 22:54:14] VERBOSE[2555] :        > [freepbx_ip]:2000: keepalive:62, keepaliveinterval:60
[2021-11-25 22:54:15] VERBOSE[3867] :     -- <?xml version="1.0" encoding="UTF-8"?>
[2021-11-25 22:54:15] VERBOSE[3867] :     -- <x-cisco-alarm>
[2021-11-25 22:54:15] VERBOSE[3867] :     -- <Alarm Name="LastOutOfServiceInformation">
[2021-11-25 22:54:15] VERBOSE[3867] :   == Alarm Type: LastOutOfServiceInformation
[2021-11-25 22:54:15] VERBOSE[3867] :     -- <ParameterList>
[2021-11-25 22:54:15] VERBOSE[3867] :     -- <String name="DeviceName">SEP[phone_mac]</String>
[2021-11-25 22:54:15] VERBOSE[3867] :     -- <String name="DeviceIPv4Address">[phone_ip]/24</String>
[2021-11-25 22:54:15] VERBOSE[3867] :     -- <String name="IPv4DefaultGateway">[gateway_ip]</String>
[2021-11-25 22:54:15] VERBOSE[3867] :     -- <String name="DeviceIPv6Address"></String>
[2021-11-25 22:54:15] VERBOSE[3867] :     -- <String name="IPv6DefaultGateway"></String>
[2021-11-25 22:54:15] VERBOSE[3867] :     -- <String name="ModelNumber">CP-7962G</String>
[2021-11-25 22:54:15] VERBOSE[3867] :     -- <String name="NeighborIPv4Address"></String>
[2021-11-25 22:54:15] VERBOSE[3867] :     -- <String name="NeighborIPv6Address"></String>
[2021-11-25 22:54:15] VERBOSE[3867] :     -- <String name="NeighborDeviceID"></String>
[2021-11-25 22:54:15] VERBOSE[3867] :     -- <String name="NeighborPortID"></String>
[2021-11-25 22:54:15] VERBOSE[3867] :     -- <Enum name="DHCPv4Status">1</Enum>
[2021-11-25 22:54:15] VERBOSE[3867] :     -- <Enum name="DHCPv6Status">0</Enum>
[2021-11-25 22:54:15] VERBOSE[3867] :     -- <Enum name="TFTPCfgStatus">1</Enum>
[2021-11-25 22:54:15] VERBOSE[3867] :     -- <Enum name="DNSStatusUnifiedCM1">4</Enum>
[2021-11-25 22:54:15] VERBOSE[3867] :     -- <Enum name="DNSStatusUnifiedCM2">0</Enum>
[2021-11-25 22:54:15] VERBOSE[3867] :     -- <Enum name="DNSStatusUnifiedCM3">0</Enum>
[2021-11-25 22:54:15] VERBOSE[3867] :     -- <String name="VoiceVLAN">4095</String>
[2021-11-25 22:54:15] VERBOSE[3867] :     -- <String name="UnifiedCMIPAddress">[freepbx_ip]</String>
[2021-11-25 22:54:15] VERBOSE[3867] :     -- <String name="LocalPort">49152</String>
[2021-11-25 22:54:15] VERBOSE[3867] :     -- <String name="TimeStamp">14845570211301484557040312148455705930014845570799061484557098707148455711751414845571364671484557155295148455717514514845571951641484557219870148455723924614845572591611484557278038148455702520614845570444541484557026394148455704596114845570647051484557084473148455710384714845571225371484557143116148455716327114845571830741484557203062148455722273914845572436521484557264323148455728482514845573046001484557324606148455734397714845573649031484557385411148455740529314845574253571484557446265148455746704114845574875171484557507736148455752648114845575457811484557566216148455758718914845576065511484557626586148455764536214845576646611484557685225148455770551014845577260071484557746972148455776653914845577863881484557807026148455782675314845578458651484557865478148455788543514845579044971484557924902148455794485214845579642291484557984861148455800353414845580239911484558044558148455806316514845580825751484558103089148455812174714845581408091484558160488148455818039114845582002051484558618014
[2021-11-25 22:54:15] VERBOSE[3867] :  SEP[phone_mac]: is registering, Instance: 0, UserId: 0, Type: Cisco 7962 (404), Version: 22 (loadinfo 'term62.default')
[2021-11-25 22:54:15] VERBOSE[3867] :     -- SCCP: Device 'SEP[phone_mac]' not found in realtime table 'sccpdevice'
[2021-11-25 22:54:15] NOTICE[3867] sccp_actions.c: SEP[phone_mac]: Rejecting device: Device Unknown 
[2021-11-25 22:54:16] VERBOSE[3867] :     -- SCCP: Exiting sccp_socket device thread
[2021-11-25 22:54:16] VERBOSE[3867] :     -- SCCP: Session without a device attached !
[2021-11-25 22:54:16] VERBOSE[3867] :     -- SCCP: cleanup session
[2021-11-25 22:54:16] VERBOSE[3867] :     -- SCCP: Destroy Session [phone_ip]
[2021-11-25 22:54:16] VERBOSE[3867] :     -- SCCP: Shutdown socket 35
[2021-11-25 22:54:16] VERBOSE[3867] :     -- SCCP: Closing socket 35
```

dkgroot, 2021-11-27:
@astifr @waffle-stomper 
It sounds like you might be using and older version of sccp_manager. You ought to be clone https://github.com/chan-sccp/sccp_manager or download it's latest release from https://github.com/chan-sccp/sccp_manager/releases/tag/v14.2.0.3. This newer version should automatically check and setup realtime for you and would make the same changes described by @waffle-stomper . The changes made by @waffle-stomper manually are fine and would also work, but should not be required anymore using the latest version.

@waffle-stomper Thanks for helping out another member of our community.

waffle-stomper, 2021-11-28:
@dkgroot 

Thanks for the quick response, and for all of your hard work on this project.

> It sounds like you might be using and older version of sccp_manager.

I think you're right - I was using Legacy-v2.tar.gz which was recommended on this page in the wiki: https://github.com/chan-sccp/chan-sccp/wiki/FreePBX_SCCP-modules

I considered v14.x but I read that it would require upgrading PHP to v7 and I was worried that might break my FreePBX install.

dkgroot, 2021-11-28:
@waffle-stomper Yes the php update/upgrade is a requirement, but if you don't already have it, i would assume your FreePBX install is in need of a little update as well. If you are running on a VM, you should be able do a bulkexport on your current install, start a new installation with updated OS and FreePBX and then proceed to use bulkimport to get everything back up and running.

Note: I am not a FreePBX user, so i am just guessing at this point ! Steve and Alex are in the know and they might be the better advisors here.

steve-lad, 2021-11-28:
FreePbx 16 is now GA, and there is an upgrade tool from 15, so the best path is to upgrade FreePbx, and then install 14.3. That will resolve any issues with the flat file configuration.

@astifr Normally you do not need a new db user - just use the freepbx user. 14.3 will take care of this

dkgroot, 2021-11-28:
@steve-lad Thanks for your guidance !
@astifr If you are running on/in a VM, it's always wise to snapshot or clone before making modifications, just so you have something to fall back on. Backup's are a good thing :-)

astifr, 2021-11-29:
Thank you all for your support. I´ll try as soon as possible.

closed 2021-11-29 by astifr