chan-sccp/chan-sccp #585: Assigning a line to a phone doesn’t work

slipdip123, 2021-11-18:
When the phone registers with chan SCCP (on FreePBX), the phone (Cisco 7962G) doesn’t seem to receive any lines or proper button layouts. On SCCP manager, the line doesn’t show up in the line column and my phone only has the buttons redial, new call and more.

**Reproduction Steps:**

1. Install chan-SCCP and SCCP manager
2. First attempt to register phones fail as the database doesn’t have anything in it/can’t be accessed, so run mysql -p asterisk < /usr/src/chan-sccp/conf/mysql-v5_enum.sql again, and add the proper db user and pass in res_config_mysql.conf
3. Create phone in SCCP manager, but the line doesn’t show
4. Phone registers with wrong buttons and no lines

**chan-sccp version:** SCCP manager 14.2.0.11 (I installed around an hour ago so should be up to date)

**asterisk version:** asterisk 16.2, FPBX 15.0.17.62

**Expected behavior:**

phone registers with a line and has correct button plan

**Observed behavior:**

Phone registers with no lines and incorrect button plan

**Addition / Optional Information:**
- Problem started happening recently, didn't happen in an older version of Chan-SCCP: No
- Problem can be reliably reproduced, doesn't happen randomly: I don’t know

slipdip123, 2021-11-18:
I fixed this by reinstalling SCCP manager through FreePBX. Don't know why it didn't work before but it works now.

closed 2021-11-18 by slipdip123

dkgroot, 2021-11-21:
@slipdip123 This was an sccp_manager issue and should ought to be reported on it's issue list, otherwise it might not receive much attention, sorry for responding a little late. Congrats on getting it working and good luck.