chan-sccp/chan-sccp #586: Keyset layouts don't work

slipdip123, 2021-11-18:
I have setup 2 Cisco 7962G phones and they keyset layouts don't seem to be applied. On hook, all I see is redial, new call and more. There is now call forward and the more button only shows things like private calling.

**Reproduction Steps:**

1. Connect Cisco phones
2. Register phones

**chan-sccp version:** SCCP manager 14.2.0.11, installed only a few hours ago

**asterisk version:** 16.2, FPBX 15.0.17.62
**Expected behavior:**

Keyset is present on both units

**Observed behavior:**

Keyset is not present on both units

**Screenshots**

[image]

[image]

slipdip123, 2021-11-18:
I am stupid once again, didn't select the correct things in the actual phone settings

closed 2021-11-18 by slipdip123

dkgroot, 2021-11-21:
Happy to see, some people can manage nice issue reports and still fend for themselves and close the ticket on the next day ;-)

For future reference, use: https://github.com/chan-sccp/sccp_manager/issues to report FreePBX + sccp_manager issues. So that the team can pick them up quicker. Also for quick questions you might wanna try out the [gitter](https://gitter.im/sccp_manager/community) chat.