chan-sccp/chan-sccp #587: Wiki contains redundant page(s) about Realtime Configuration

waffle-stomper, 2021-11-29:
The wiki contains two very similar pages, both of which might be unnecessary now that SCCP Manager takes care of the Realtime DB setup:

https://github.com/chan-sccp/chan-sccp/wiki/Realtime-Configuration

https://github.com/chan-sccp/chan-sccp/wiki/using-realtime-database

dkgroot, 2022-01-29:
@waffle-stomper 
- sccp_manager users should be made aware that they should not do any of this (maybe at the top of the page), as sccp_manager will take care of this
- Would you be able and willing to merge these two pages into one (picking the best information from the two) ? Most of the time users are must better at writing helpfull wiki pages then developers are
- After that would have to do a quick search for any links to either page and replace them with the merged one.

waffle-stomper, 2022-02-05:
- Both pages merged into https://github.com/chan-sccp/chan-sccp/wiki/Realtime-Configuration
- Note added to the top of the page to warn new users
- Links updated

closed 2022-02-05 by waffle-stomper

dkgroot, 2022-02-24:
@waffle-stomper Thank you very much !