chan-sccp/chan-sccp #588: Resume a call take to Asterisk crashing

choanhemnhe, 2021-11-30:
Hello,
I caught the following problem in current SCCP driver. (SCCP 4.3.4 develop - 52ed8b2)

Here is the story.
If you are on the phone with the first call, then the second call coming. You answer the second call, the first call be putted on hold. Now you finished the second call, then Resume the first call. At this moment the Asterisk will crash. This will occur for the CISCO 7940 only.

Follow is the steps to re-product the problem.
Suppose you have 2 extension, 100 and 101 on the CISCO 7940s
1.	Use cellphone to call 100 then answer the call.
2.	Now use 101 to call 100, answer the call, the call with cellphone will be putted on hold.
3.	Next finish the call with 100.
4.	On 100 Resume the call with Cellphone, this point the Asterisk will crash.
 
I tested on Cisco 7941 and Cisco 7945, they worked good, No crashing
 
I believe It will crash with Cisco 7940s only.
 
Versions:
SCCP 4.3.4 develop - 52ed8b2
Asterisk 13.18.4
 
 
Any advises are very welcome.
 
Thank a lot.

dkgroot, 2022-01-29:
@choanhemnhe Sorry it took so long to get to your issue ticket. My new job has been taking it's toll. Interesting issue, which i have not run into before.
Can you still reproduce this issue, if so could you provide [a backtrace](https://github.com/chan-sccp/chan-sccp/wiki/Reporting-Bugs) of the crash and a bit of debug log from the console (with 'sccp debug core, channel, indicate, pbx' set, beforehand). I currently do not have a 7940 available to me for direct testing, but with the information i should be able to make short work of this.

choanhemnhe, 2023-02-01:
Hi Diederik,
We are willing to pay to any contributor, who going to help us to fix this bug. Could you please help us with this, or if you can refer any one to us. We tested the latest version SCCP 4.3.5 on Asterisk 18.13.0, the problem is still there. The phone models cause the crash are Cisco 7940 and Cisco 7925. The process to re-product the crash as follow; Register the Cisco 7940, make oubound call to landline or Cell phone number, answer the call from the cellphone or landline phone. Then on the Cisco 7940, put the call on hold, then resume the call. This point Asterisk will crash. Please let me know if you or someone else can help me to fix this bug.
Thank you.