chan-sccp/chan-sccp #591: SPA514 SCCP interrupted calls by re-registrations

Raikko, 2022-02-10:
Some of Cisco SPA514G phones have problems with SCCP registrations. They are re-registering 4x per hour, once per minute. It iterrupts calls. This only happens if PC is connected to PC port.

**Reproduction Steps:**

1. Configure switch with CDP enabled, port in trunk mode, native access vlan for data and voice vlan for phone
2. Configure provisioning for phone
3. Configure sccp device and line in asterisk
4. Connect and register a SPA514 phone
5. Connect PC to PC port
6. Make a call and wait for next SCCP registration
7. It fails SCCP token refresh and it interrupts the call

**chan-sccp version:** chan-sccp-b-develop-20201114

**asterisk version:** 16.19.0

**SPA514G firmware:** 7.6.1

**Expected behavior:**

I expect that token refresh will not cause re-registration and will not drop calls.

**Observed behavior:**

Phone VLANs are set via CDP. Port is in trunk mode with access native VLAN 1 and voice (operational) VLAN 100. If phone is rebooted, it ends in reboot loop due to VLAN changing, but may not be an issue in common. If PC port is disconnected and phone rebooted, SCCP works without problem.

**Addition / Optional Information:**
- Problem started happening recently, didn't happen in an older version of Chan-SCCP: Yes (OK in chan-sccp-b-develop-5ad78e409fab3288f901c62dca9d8b33273b43de ~May 2016)
- Problem can be reliably reproduced, doesn't happen randomly: No. It seems it depends on network configuration (CDP+native VLAN)
- Platform:
  - QEMU VM
  - Intel Xeon Processor (Skylake, IBRS)
  - 64 Bit
  - Linux
  - Own distro based on CentOS 7
- Direct Contact information: Matrix @sadam:linuxbox.cz
- Log Files: https://storage.linuxbox.cz/index.php/s/HZoTTfX5EHbY4oM
- Comments: I shared asterisk sccp debug log (sccp-debug.log, GMT+1), phone syslog with +-1 hour around token refresh (spa514-192.168.67.16.log, GMT+0), phone XML configuration (SEP10bd18b01ed1.xml) and packet dump of token refresh (sccp-192.168.67.14.pcap)

**BTW: Consider making a donation via paypal:**
- https://www.paypal.com/cgi-bin/webscr?&cmd=_donations&business=chan.sccp.b.pp%40gmail.com

2br-2b, 2024-08-08:
Did you end up finding a solution or workaround for this problem?

Raikko, 2024-08-10:
Hi,no, I was not able to solve the issue.I have switched to SIP protocol. It solved problems with re-registrations. But problem with boot loops when some device was connected to PC port persist.