chan-sccp/chan-sccp #592: Error cannot compile chan-sccp-b under Asterisk 18

stergios1, 2022-02-25:
issuing make command returns with an error

**Reproduction Steps:**

1. git clone https://github.com/chan-sccp/chan-sccp.git chan-sccp
2. cd chan-sccp
3. ./configure --enable-debug --enable-conference --disable-optimization

**chan-sccp version:** 4.3.5

**asterisk version:** 18.10.0

**Expected behavior:**

make should be able to finish and compile the source code

**Observed behavior:**

  CC       libsccp_la-sccp_management.lo
  CC       libsccp_la-sccp_mwi.lo
In file included from pbx_impl/ast/ast.h:26,
                 from pbx_impl/pbx_impl.h:18,
                 from common.h:61,
                 from sccp_mwi.c:10:
sccp_mwi.c: In function 'pbxMailboxSubscribe':
pbx_impl/ast/define.h:237:38: error: implicit declaration of function 'ast_mwi_subscribe_pool'; did you mean 'pbx_mwi_subscribe_pool'? [-Werror=implicit-function-declaration]
 #define pbx_mwi_subscribe_pool       ast_mwi_subscribe_pool
                                      ^~~~~~~~~~~~~~~~~~~~~~
sccp_mwi.c:166:49: note: in expansion of macro 'pbx_mwi_subscribe_pool'
  pbx_subscription = (pbx_event_subscription_t *)pbx_mwi_subscribe_pool(subscription->mailbox->uniqueid, pbx_mwi_event, subscription);
                                                 ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.
cc1: some warnings being treated as errors
make[3]: *** [Makefile:971: libsccp_la-sccp_mwi.lo] Error 1
make[3]: Leaving directory '/usr/src/chan-sccp_master/src'
make[2]: *** [Makefile:1109: all-recursive] Error 1
make[2]: Leaving directory '/usr/src/chan-sccp_master/src'
make[1]: *** [Makefile:668: all] Error 2
make[1]: Leaving directory '/usr/src/chan-sccp_master/src'
make: *** [Makefile:615: all-recursive] Error 1

**Screenshots**

(Screenshots which show reproduction steps and/or demonstrate the problem)
[url]

**Addition / Optional Information:**
- Problem started happening recently, didn't happen in an older version of Chan-SCCP: [No] same is true for 4.3.4
- Problem can be reliably reproduced, doesn't happen randomly: [Yes]
- Platform:
  32-bit Operating System : Linux
          Kernel : 5.10.63-v7+
    Architecture : armv7l
Asterisk Version : 11900 / 119
  Asterisk Repos : TRUNK
      Build user : root
         Version : 4.3.5_develop_rc1533a6b
Compiler Version : 8
              CC : gcc
             CPP : gcc -E
         HOST_CC : gcc
          CFLAGS : -Og -Wall -I. -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -std=gnu11 -fgnu89-inline  -isystem /usr/include -fsigned-char -I/usr/include/libxml2 -g3 -g3 -ggdb3 -pthread    
        CPPFLAGS : -U_FORTIFY_SOURCE  -isystem /usr/include -fsigned-char -I/usr/include/libxml2 -g3  -g3 -ggdb3     
         LDFLAGS :  -lssl -lcrypto -lpthread -lc -lltdl  -lbfd -ldl      -L/usr/lib         
        PBX_TYPE : Asterisk
      PBX_PREFIX : 
         PBX_ETC : /etc/asterisk
         PBX_LIB : /usr/lib
      PBX_VARLIB : /var/lib/asterisk
     PBX_DATADIR : /var/lib/asterisk
     PBX_INCLUDE : /usr/include
Module Directory : /usr/lib/asterisk/modules

- Direct Contact information: [stergios1@gmail.com]
- Log Files (attach .txt / [pastbin](http://pastebin.com/) / [gist](http://gist.github.com/)): [url]
- Comments: [Other Information]

**BTW: Consider making a donation via paypal:**
- https://www.paypal.com/cgi-bin/webscr?&cmd=_donations&business=chan.sccp.b.pp%40gmail.com

dkgroot, 2022-02-26:
@stergios1
The reason is faulty version detection by the current configure script when asterisk-19 support was added. I will try to resolve this tomorrow (might have to undo the asterisk-19 support for the time being).

rjaiswal, 2022-03-26:
@dkgroot  Trying to install on asterisk 18.8.0.  Also getting same error.  any update?

jeremyy294, 2022-04-09:
Also experiencing same error when compiling on Asterisk 16.25.1

P110, 2022-06-20:
Still experiencing this issue when compiling for Asterisk 16.1.1 - is there any word on this? Thank you

P110, 2022-06-20:
It's an issue with Asterisk 19.X version checking introduced in chan-sccp v4.3.5, a workaround in the meantime is to use chan-sccp v4.3.4 (https://github.com/chan-sccp/chan-sccp/releases/tag/v4.3.4)

Can't use the latest dev branch but should hopefully get you up and running, thank you @dkgroot & contributors this is really cool functionality <3

coobic, 2022-11-17:
Also experiencing same error when compiling on Asterisk 16.27.
We temporarily changed the line 25805 in configure from:
`
$EGREP "8.0.0" >/dev/null 2>&1; then :
`
to
`
$EGREP "181.0.0" >/dev/null 2>&1; then :
`

for skip Asterisk 19 version checks.

This is not a solution, but a crutch. If you really want to, you can temporarily use it for compile the project.

rjaiswal, 2023-05-18:
> Also experiencing same error when compiling on Asterisk 16.27. We was temporarily changed the line 25805 in configure from: `$EGREP "8.0.0" >/dev/null 2>&1; then :` to `$EGREP "181.0.0" >/dev/null 2>&1; then :`
> 
> for skip Asterisk 19 version checks.
> 
> This is not a solution, but a crutch. If you really want to, you can temporarily use it for compile the project.

This worked for me.