chan-sccp/chan-sccp #595: error while compiling chan-sccp with asterisk-19.3

subware, 2022-04-01:
compile-error "sccp_mwi.c: In function ‘pbxMailboxSubscribe’:
pbx_impl/ast/define.h:237:38: error: implicit declaration of function ‘ast_mwi_subscribe_pool’; did you mean ‘pbx_mwi_subscribe_pool’? [-Werror=implicit-function-declaration]
  237 | #define pbx_mwi_subscribe_pool       ast_mwi_subscribe_pool

**Reproduction Steps:**

1. git clone actual version chan-sccp
2. according to manual:
3.  ./configure (ok and without any errors)
4. make cannot finish - error see below:
[image]

**chan-sccp version:** (include revision and/or branch if relevant)
Version-Info:
[image]

**asterisk version:** 19.3

**Expected behavior:** compile finish without any errors to go to next steps

**Screenshots**

(Screenshots which show reproduction steps and/or demonstrate the problem)
[url]

- Direct Contact information: [You can contact me here]

- Comments: if you need further logs / screenshots or anything else, pls let me know.

sagaciouszu, 2022-06-06:
Also encountering this error on IncrediblePBX 2021/Asterisk 19 on armhf/Pi4. 

`In file included from pbx_impl/ast/ast.h:26,
                 from pbx_impl/pbx_impl.h:18,
                 from common.h:61,
                 from sccp_mwi.c:10:
sccp_mwi.c: In function 'pbxMailboxSubscribe':
pbx_impl/ast/define.h:237:38: error: implicit declaration of function 'ast_mwi_subscribe_pool'; did you mean 'pbx_mwi_subscribe_pool'? [-Werror=implicit-function-declaration]
 #define pbx_mwi_subscribe_pool       ast_mwi_subscribe_pool
                                      ^~~~~~~~~~~~~~~~~~~~~~
sccp_mwi.c:166:49: note: in expansion of macro 'pbx_mwi_subscribe_pool'
  pbx_subscription = (pbx_event_subscription_t *)pbx_mwi_subscribe_pool(subscription->mailbox->uniqueid, pbx_mwi_event, subscription);
                                                 ^~~~~~~~~~~~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.
cc1: some warnings being treated as errors
make[3]: *** [Makefile:971: libsccp_la-sccp_mwi.lo] Error 1
make[3]: *** Waiting for unfinished jobs....
make[3]: Leaving directory '/usr/src/chan-sccp/src'
make[2]: *** [Makefile:1109: all-recursive] Error 1
make[2]: Leaving directory '/usr/src/chan-sccp/src'
make[1]: *** [Makefile:668: all] Error 2
make[1]: Leaving directory '/usr/src/chan-sccp/src'
make: *** [Makefile:615: all-recursive] Error 1`

`Operating System : Linux
          Kernel : 5.10.103-v7l+
    Architecture : armv7l
Asterisk Version : 11900 / 119
  Asterisk Repos : TRUNK
      Build user : someonesexy
         Version : 4.3.5_develop_rc1533a6b
Compiler Version : 8
              CC : gcc
             CPP : gcc -E
         HOST_CC : gcc
          CFLAGS : -Og -Wall -I. -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -std=gnu11 -fgnu89-inline  -isystem /usr/include -fsigned-char -I/usr/include/libxml2 -g3 -g3 -ggdb3 -pthread
        CPPFLAGS : -U_FORTIFY_SOURCE  -isystem /usr/include -fsigned-char -I/usr/include/libxml2 -g3  -g3 -ggdb3
         LDFLAGS :  -lssl -lcrypto -lpthread -lc -lltdl  -lbfd -ldl      -L/usr/lib
        PBX_TYPE : Asterisk
      PBX_PREFIX :
         PBX_ETC : /etc/asterisk
         PBX_LIB : /usr/lib
      PBX_VARLIB : /var/lib/asterisk
     PBX_DATADIR : /var/lib/asterisk
     PBX_INCLUDE : /usr/include
Module Directory : /usr/lib/asterisk/modules`

sagaciouszu, 2022-06-06:
Dove into the closed issues and found the answer.

```
make install-headers
```

From your /usr/src/asterisk-x.x.x/ before a `make`.

closed 2026-10-30 by subware