chan-sccp/chan-sccp #609: Deprecated functions in Asterisk v21

2br-2b, 2024-02-15:
While I can't find the exact documentation, ast_channel_macroexten is deprecated in Asterisk v21. The sources I have for this are I-tried-it and this commit: https://github.com/asterisk/asterisk/commit/e8f548c1557620df8a7231cbede6cb9b052d352b

I haven't confirmed that it still works on v20, but it looks like that commit is only in the v21 tags, so I assume it still works fine there

**Reproduction Steps:**

1. Try to install chan-sccp on Asterisk v21
2. Get an error

**chan-sccp version:** v4.3.5

**asterisk version:** v21+

lachesis, 2024-02-17:
Confirmed, doesn't build for me on Asterisk 21 either. I guess the right answer is to remove all of the macro related code?

lachesis, 2024-02-17:
It built for me with this patch:
```
diff --git a/src/pbx_impl/ast116/ast116.c b/src/pbx_impl/ast116/ast116.c
index c162e755..f6364dca 100644
--- a/src/pbx_impl/ast116/ast116.c
+++ b/src/pbx_impl/ast116/ast116.c
@@ -3065,10 +3065,6 @@ DECLARE_PBX_CHANNEL_STRGET(name)
     DECLARE_PBX_CHANNEL_STRGET(linkedid)
     DECLARE_PBX_CHANNEL_STRGET(context)
     DECLARE_PBX_CHANNEL_STRSET(context)
-    DECLARE_PBX_CHANNEL_STRGET(macroexten)
-    DECLARE_PBX_CHANNEL_STRSET(macroexten)
-    DECLARE_PBX_CHANNEL_STRGET(macrocontext)
-    DECLARE_PBX_CHANNEL_STRSET(macrocontext)
     DECLARE_PBX_CHANNEL_STRGET(call_forward)
     DECLARE_PBX_CHANNEL_STRSET(call_forward)

@@ -3536,10 +3532,6 @@ const PbxInterface iPbx = {
        setChannelExten: sccp_astwrap_set_channel_exten,
        getChannelContext: sccp_astwrap_get_channel_context,
        setChannelContext: sccp_astwrap_set_channel_context,
-       getChannelMacroExten: sccp_astwrap_get_channel_macroexten,
-       setChannelMacroExten: sccp_astwrap_set_channel_macroexten,
-       getChannelMacroContext: sccp_astwrap_get_channel_macrocontext,
-       setChannelMacroContext: sccp_astwrap_set_channel_macrocontext,
        getChannelCallForward: sccp_astwrap_get_channel_call_forward,
        setChannelCallForward: sccp_astwrap_set_channel_call_forward,

@@ -3687,10 +3679,6 @@ const PbxInterface iPbx = {
        .setChannelExten = sccp_astwrap_set_channel_exten,
        .getChannelContext = sccp_astwrap_get_channel_context,
        .setChannelContext = sccp_astwrap_set_channel_context,
-       .getChannelMacroExten = sccp_astwrap_get_channel_macroexten,
-       .setChannelMacroExten = sccp_astwrap_set_channel_macroexten,
-       .getChannelMacroContext = sccp_astwrap_get_channel_macrocontext,
-       .setChannelMacroContext = sccp_astwrap_set_channel_macrocontext,
        .getChannelCallForward = sccp_astwrap_get_channel_call_forward,
        .setChannelCallForward = sccp_astwrap_set_channel_call_forward,
```
The module that it build loaded. I have not tested further yet. Will report back once my first SCCP phone is here and I test it out later today.

endor3821, 2025-01-01:
has there been any development on this issue. perhaps a ./configure --disable-macros feature

2br-2b, 2025-01-02:
I don't believe the original contributors are still active at this time. If you build from the PR branch, that should build with the proposed changes

endor3821, 2025-01-08:
So after a lot of research, I have discovered the issue. The sccp protocol is being phased out. There may be many reasons for this, ie cisco complaints against open source use of it. Asterisks >19 has dropped the chan_skinny and ast_macro from the official build. So the only way to truly fix this is to recompile asterisk with these modules. This is a daunting task seeing has it most likely with break other portions of the code body. So it wont be long until these devices are relegated to the past.

fhloston, 2025-01-15:
so far chan-sccp also compiles with asterisk 22 with these [1] changes. phones register fine.

however, during an attempted call from pjsip i do get the error "Don't know how to send format alaw packets with RTP" from this issue [2].

same for the echotest application.

@lachesis have you actually been able to route calls  to chan_sccp devices/channels? what am i missing?
 
[1] https://github.com/chan-sccp/chan-sccp/issues/609#issuecomment-1950275587
[2] https://github.com/chan-sccp/chan-sccp/issues/617

UBUNTUGOD, 2025-10-15:
@MadManChLP same i have a set of 7965's and will register for a frew minutes then not register anymore (using SIP)