chan-sccp/chan-sccp #617: Don't know how to send format alaw packets with RTP

CRCinAU, 2024-12-24:
It seems with newer versions of Asterisk, chan_sccp-b will fail a call setup. eg #615

> core show version 
Asterisk 20.10.0 built by $me @ $server on a x86_64 running Linux on 2024-10-17 16:05:45 UTC

Log during a call attempt:
```
    -- Executing [998@phones:1] Answer("SCCP/101-00000003", "") in new stack
       > 0x7fbdc000f920 -- Strict RTP learning after remote address set to: 172.31.1.196:18262
       > 0x7fbdc000f920 -- Strict RTP switching to RTP target address 172.31.1.196:18262 as source
    -- Executing [998@phones:2] MusicOnHold("SCCP/101-00000003", "") in new stack
    -- Started music on hold, class 'default', on channel 'SCCP/101-00000003'
[Dec 25 00:32:38] WARNING[16205][C-00000004]: res_rtp_asterisk.c:5532 ast_rtp_write: Don't know how to send format alaw packets with RTP
[Dec 25 00:32:38] WARNING[16205][C-00000004]: res_musiconhold.c:1097 moh_generate: Failed to write frame to 'SCCP/101-00000003': Resource temporarily unavailable
    -- Stopped music on hold on SCCP/101-00000003
  == Spawn extension (phones, 998, 2) exited non-zero on 'SCCP/101-00000003'
```

Asterisk codec translations:
```
voip*CLI> core show translation 
         Translation times between formats (in microseconds) for one second of data
          Source Format (Rows) Destination Format (Columns)

           ulaw  alaw   gsm slin8 slin12 slin16 slin24 slin32 slin44 slin48 slin96 slin192  g722  opus
     ulaw     - 15000 15000  9000  17000  17000  17000  17000  17000  17000  17000   17000 17250 23000
     alaw 15000     - 15000  9000  17000  17000  17000  17000  17000  17000  17000   17000 17250 23000
      gsm 15000 15000     -  9000  17000  17000  17000  17000  17000  17000  17000   17000 17250 23000
    slin8  6000  6000  6000     -   8000   8000   8000   8000   8000   8000   8000    8000  8250 14000
   slin12 14500 14500 14500  8500      -   8000   8000   8000   8000   8000   8000    8000 14000 14000
   slin16 14500 14500 14500  8500   8500      -   8000   8000   8000   8000   8000    8000  6000 14000
   slin24 14500 14500 14500  8500   8500   8500      -   8000   8000   8000   8000    8000 14500 14000
   slin32 14500 14500 14500  8500   8500   8500   8500      -   8000   8000   8000    8000 14500 14000
   slin44 14500 14500 14500  8500   8500   8500   8500   8500      -   8000   8000    8000 14500 14000
   slin48 14500 14500 14500  8500   8500   8500   8500   8500   8500      -   8000    8000 14500  6000
   slin96 14500 14500 14500  8500   8500   8500   8500   8500   8500   8500      -    8000 14500 14500
  slin192 14500 14500 14500  8500   8500   8500   8500   8500   8500   8500   8500       - 14500 14500
     g722 15600 15600 15600  9600  17500   9000  17000  17000  17000  17000  17000   17000     - 23000
     opus 23500 23500 23500 17500  17500  17500  17500  17500  17500   9000  17000   17000 23500     -
voip*CLI>
```

It doesn't seem to matter what codec I use for the chan_sccp-b phone... Have tried both g722 and alaw.

Codec related sccp.conf values:
```
[general]
servername = Asterisk
keepalive = 45
debug = 0
dateFormat = D/M/Y
bindaddr = ::
language=en_GB
;bindaddr = 0.0.0.0
port = 2000
disallow=all
allow=g722,alaw,ulaw
...
[defaultdevice](!)                      ; default device template
type = device                           ; specifies that this template is for a device, it will be inherited
transfer = on                           ; enable or disable the transfer capability. It does remove the transfer softkey
park = off                              ; take a look to the compile howto. Park stuff is not compiled by default
cfwdall = on                            ; activate the callforward stuff and softkeys
cfwdbusy = on
cfwdnoanswer = on
dndFeature = on
earlyrtp = progress                     ; valid options: none, offhook, dial, ringout and progress. default is progress.
                                        ; The audio stream will be open in the progress and connected state by default.
private = on                            ; permit the private function softkey for this device
mwilamp = on                            ; Set the MWI lamp style when MWI active to on, off, wink, flash or blink
mwioncall = on                          ; Set the MWI on call.
softkeyset = softkeyset                 ; use softkeyset with name softkeyte new template named 7960
privacy = on                            ; full = disable hints notification on devices, on = hints showed depending on private key, off = hints always showed
;force_dtmfmode = rfc2833               ; auto, skinny or rfc2833. Some phone models with bad firmware do send dtmf in a messed up order and need to be forced to skinny mode.
```

sgofferj, 2025-01-19:
Yeah, that was introduced somewhen between 18.7 and 18.25... I just upgraded and I'm running into the same problem

linus0508, 2025-02-15:
Was anyone able to elaborate on that further? Im currently experiencing the same bug but neither earlier versions (tried 18.4) or newer versions (18.17) of Asterisk are working for me. Always the same error as described above.
All input is greatly appreciated!

CRCinAU, 2025-03-02:
I'm just having a hunt.....

Compiling from `asterisk-20.9.3.tar.gz` seems to work... Having a hunt to see what version first breaks....

mpaland, 2025-03-10:
Just came back here to check the project status.
I'm on 18.19.0 for a while now without any problems. And... I pretty much think that I'm gonna be stranded here for a now.
No other phone in the world can replace the good old 797x series. Newer Cisco phones look and feel like plastic toys. Have no advice what to do.

sgofferj, 2025-03-10:
> Just came back here to check the project status. I'm on 18.19.0 for a while now without any problems. And... I pretty much think that I'm gonna be stranded here for a now. No other phone in the [...]

For that reason, I bit the sour apple and changed my 7960 and 7970 to SIP firmware. I'm also probably going to replace my 7960 with 7970 soon. 7970 SIP is a hassle to configure in comparison to SCCP but once it's done, it seems to work OK.

nikolabura, 2025-03-17:
Removing the following section in Asterisk's `main/rtp_engine.c` seems to fix the issue for me:
```c
ast_rwlock_rdlock(&codecs->codecs_lock);
if (payload >= 0 && payload < AST_VECTOR_SIZE(&codecs->payload_mapping_tx)){
	type = AST_VECTOR_GET(&codecs->payload_mapping_tx, payload);
	if (!type || (sample_rate != 0 && type->sample_rate != sample_rate)) {
		/* Don't use the type if we can't find it or it doesn't match the supplied sample_rate */
		payload = -1;
	}
}
ast_rwlock_unlock(&codecs->codecs_lock);
```
It was added in [this commit](https://github.com/asterisk/asterisk/commit/055031dfcb669cb84684565db7a1fdc7073d6d79).

(I'm not an Asterisk dev, but at a glance, it looks like `find_static_payload_type` pulls from `static_RTP_PT`, and then the above code uses it to index into `codecs->payload_mapping_tx`, which seems to be a vector full of nulls when chan_sccp is in use, at least at the time that function runs.)

mpaland, 2025-04-09:
> For that reason, I bit the sour apple and changed my 7960 and 7970 to SIP firmware. I'm also probably going to replace my 7960 with 7970 soon. 7970 SIP is a hassle to configure in comparison to SCCP but once it's done, it seems to work OK.

@sgofferj, I never tried the SIP images as I read a lot of complains about stability issues. What is your experience? Is the BLF working fine?

sgofferj, 2025-04-09:
> [@sgofferj](https://github.com/sgofferj), I never tried the SIP images as I read a lot of complains about stability issues. What is your experience? Is the BLF working fine?

I'm not using BLF but speeddials on 7970/7914 work fine. 7960 doesn't support 7914 on SIP firmware.
Normal telephony operations have been without problems so far.

2br-2b, 2025-04-26:
Is there a good guide somewhere on how to easily migrate SCCP phones to SIP? That would be a huge help to us over here.

sgofferj, 2025-04-27:
@2br-2b Don't remember. I think on voip-info.org was one.

scientress, 2025-04-30:
I can confirm that removing the code block from asterisk @nikolabura suggested fixes the error. That's of course not a permanent solution but a working hot fix.

If you are on gentoo you can easily accomplish this by placing the attached patch in `/etc/portage/patches/net-misc/asterisk/fix-chan-sccp.patch`

```
diff --git a/main/rtp_engine.c b/main/rtp_engine.c
index 1bf76008e5..bc8e4e1807 100644
--- a/main/rtp_engine.c
+++ b/main/rtp_engine.c
@@ -2130,16 +2130,6 @@ int ast_rtp_codecs_payload_code_tx_sample_rate(struct ast_rtp_codecs *codecs, in
 		ast_rwlock_rdlock(&static_RTP_PT_lock);
 		payload = find_static_payload_type(asterisk_format, format, code);
 		ast_rwlock_unlock(&static_RTP_PT_lock);
-
-		ast_rwlock_rdlock(&codecs->codecs_lock);
-		if (payload >= 0 && payload < AST_VECTOR_SIZE(&codecs->payload_mapping_tx)){
-			type = AST_VECTOR_GET(&codecs->payload_mapping_tx, payload);
-			if (!type || (sample_rate != 0 && type->sample_rate != sample_rate)) {
-				/* Don't use the type if we can't find it or it doesn't match the supplied sample_rate */
-				payload = -1;
-			}
-		}
-		ast_rwlock_unlock(&codecs->codecs_lock);
 	}
 
 	return payload;
```

Tested with asterisk 20.12.0

pavelkim, 2025-11-15:
Asterisk 18.26.4
chan_sccp 4.3.5 TRUNK - 90dc24f

all freshly built, failing the same way:

```
    -- SEP001BD40B6C90 - SCCP/50101-0000001E: (recalculateAudioCodecformat) 
	channel capabilities: [slin16, ulaw/64k, alaw/64k, g729b, g729ab, g729, g729a]
	shared preferences [g729, g729a, g729b, g729ab]
	line preferences: [g729, g729a, g729b, g729ab, g729/annex/b]
	remote caps: [g729]
	Format:g729
    -- SCCP/50101-0000001E, OpenReceiveChannel with format G.729, payload 11, echocancel: YES, passthrupartyid: 4294967265, pbx_channel_name: SCCP/50101-0000001E

...

[Nov 15 04:29:33] WARNING[300955][C-00000016]: res_rtp_asterisk.c:5593 ast_rtp_write: Don't know how to send format g729 packets with RTP
    -- Channel SCCP/50101-0000001E left 'native_rtp' basic-bridge <2d83d8c6-85e7-43fc-99af-040a21989464>
    -- Channel SCCP/50102-0000001F left 'native_rtp' basic-bridge <2d83d8c6-85e7-43fc-99af-040a21989464>
```

fhloston, 2026-05-24:
Thank you so much for that. This also works for me.