chan-sccp/chan-sccp #618: Build chan_sccp with Freepbx 17/Asterisk 21

infineonnetworks, 2024-12-31:
Anyone able to even build chan_sccp with Asterisk 21/Freepbx 17? 
It even fails at configuration stage of not finding asterisk itself, even when pointing the configure path to the asterisk installation. 

**Reproduction Steps:**

1. Install Debian and Freepbx 17 from script as instructed
2. Pull chan_sccp code from git and try to build.

chan-sccp version: latest  v4.3.5

asterisk version: Asterisk 21

**Expected behavior:**
Expected to build and even find asterisk installation, even with PATH provided in config

**Observed behavior:**
Configure can never find Asterisk no matter what path is provided.

2br-2b, 2024-12-31:
Hello! Can you provide the specific error you're seeing? This might be related to #609

infineonnetworks, 2025-01-15:
System that was installed is debian 12 basic, then following the freepbx instructions to install freepbx 17.
Sources are in /usr/src for all chan_sccp & FreePBX. 

Configure with variables where run is as following:
./configure  --enable-conference --enable-advanced-functions --enable-distributed-devicestate --enable-video

It seems that it cant even find the asterisk installed version as per output:

Checking Configure Options:
checking pkg-config asterisk... not found
checking Search Path: /usr /usr/local /opt... not-found
not-found
not-found
configure: Either install asterisk and asterisk-devel packages using your package manager.
configure: If you compiled asterisk manually, make sure you also run `make install-headers` so chan-sccp can find them.
configure: Or run ./configure --with-asterisk=PATH with PATH pointing to the directory where you installed asterisk
configure: error: Could not find pbx headers or libraries - these are required.

FreePBX#>:/usr/src/chan-sccp# which asterisk
/usr/sbin/asterisk

FreePBX#:/usr/src/chan-sccp# asterisk -rvvv
Asterisk 21.5.0, Copyright (C) 1999 - 2022, Sangoma Technologies Corporation and others.
Created by Mark Spencer <markster@digium.com>
Asterisk comes with ABSOLUTELY NO WARRANTY; type 'core show warranty' for details.
This is free software, with components licensed under the GNU General Public
License version 2 and other licenses; you are welcome to redistribute it under
certain conditions. Type 'core show license' for details.
=========================================================================
Connected to Asterisk 21.5.0 currently running on FreePBX (pid = 1798)

Not sure if there are some environment variables that may need to be set or not. (If so it may be worth while to be in documentation)

garrykl70, 2025-01-30:
Operating System : Linux
          Kernel : 6.1.0-30-amd64
    Architecture : x86_64
Asterisk Version : 11900 / 119
  Asterisk Repos : TRUNK
      Build user : root
         Version : 4.3.5_develop_r90dc24f2
Compiler Version : 12
              CC : gcc
             CPP : gcc -E
         HOST_CC : gcc
          CFLAGS : -Og -Wall -I. -D_LARGEFILE_SOURCE -D_FILE_OFFSET_BITS=64 -std=gnu11 -fgnu89-inline  -isystem /usr/include -I/usr/include/libxml2 -g3 -g3 -ggdb3
        CPPFLAGS : -U_FORTIFY_SOURCE  -isystem /usr/include -I/usr/include/libxml2 -g3  -g3 -ggdb3
         LDFLAGS :  -lssl -lcrypto -lpthread -lc -lltdl  -lbfd -ldl      -L/usr/lib
        PBX_TYPE : Asterisk
      PBX_PREFIX :
         PBX_ETC : /etc/asterisk
         PBX_LIB : /usr/lib
      PBX_VARLIB : /var/lib/asterisk
     PBX_DATADIR : /var/lib/asterisk
     PBX_INCLUDE : /usr/include
Module Directory : /usr/lib/asterisk/modules
root@TVE-VoIP:/usr/src/chan-sccp# make
Making all in src
make[1]: вход в каталог «/usr/src/chan-sccp/src»
make  all-recursive
make[2]: вход в каталог «/usr/src/chan-sccp/src»
Making all in pbx_impl
make[3]: вход в каталог «/usr/src/chan-sccp/src/pbx_impl»
  CC       pbx_impl.lo
  CCLD     libpbximpl.la
make[3]: выход из каталога «/usr/src/chan-sccp/src/pbx_impl»
Making all in pbx_impl/ast
make[3]: вход в каталог «/usr/src/chan-sccp/src/pbx_impl/ast»
  CC       libast_la-ast.lo
  CCLD     libast.la
make[3]: выход из каталога «/usr/src/chan-sccp/src/pbx_impl/ast»
Making all in pbx_impl/ast119/
make[3]: вход в каталог «/usr/src/chan-sccp/src/pbx_impl/ast119»
  CC       libast119_la-ast119.lo
ast119.c: In function ‘sccp_astwrap_get_channel_macroexten’:
ast119.c:3046:24: error: implicit declaration of function ‘ast_channel_macroexten’; did you mean ‘pbx_channel_macroexten’? [-Werror=implicit-function-declaration]
 3046 |                 return ast_channel_##_field(channel->owner);                                                    \
      |                        ^~~~~~~~~~~~
ast119.c:3068:5: note: in expansion of macro ‘DECLARE_PBX_CHANNEL_STRGET’
 3068 |     DECLARE_PBX_CHANNEL_STRGET(macroexten)
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.
cc1: some warnings being treated as errors
make[3]: *** [Makefile:581: libast119_la-ast119.lo] Ошибка 1
make[3]: выход из каталога «/usr/src/chan-sccp/src/pbx_impl/ast119»
make[2]: *** [Makefile:1109: all-recursive] Ошибка 1
make[2]: выход из каталога «/usr/src/chan-sccp/src»

thijsvdc, 2025-05-05:
Same issue here using FreePBX 17 with Astersk 22.3.0:

`~/chan-sccp# make
Making all in src
make[1]: Entering directory '/root/chan-sccp/src'
make  all-recursive
make[2]: Entering directory '/root/chan-sccp/src'
Making all in pbx_impl
make[3]: Entering directory '/root/chan-sccp/src/pbx_impl'
make[3]: Nothing to be done for 'all'.
make[3]: Leaving directory '/root/chan-sccp/src/pbx_impl'
Making all in pbx_impl/ast
make[3]: Entering directory '/root/chan-sccp/src/pbx_impl/ast'
make[3]: Nothing to be done for 'all'.
make[3]: Leaving directory '/root/chan-sccp/src/pbx_impl/ast'
Making all in pbx_impl/ast119/
make[3]: Entering directory '/root/chan-sccp/src/pbx_impl/ast119'
  CC       libast119_la-ast119.lo
ast119.c: In function ‘sccp_astwrap_set_channel_name’:
ast119.c:3055:17: error: implicit declaration of function ‘pbx_channel_name_set’; did you mean ‘ast_channel_name_set’? [-Werror=implicit-function-declaration]
 3055 |                 pbx_channel_##_field##_set(channel->owner, _field);                                             \
      |                 ^~~~~~~~~~~~
ast119.c:3060:5: note: in expansion of macro ‘DECLARE_PBX_CHANNEL_STRSET’
 3060 |     DECLARE_PBX_CHANNEL_STRSET(name)
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.
cc1: some warnings being treated as errors
make[3]: *** [Makefile:581: libast119_la-ast119.lo] Error 1
make[3]: Leaving directory '/root/chan-sccp/src/pbx_impl/ast119'
make[2]: *** [Makefile:1109: all-recursive] Error 1
make[2]: Leaving directory '/root/chan-sccp/src'
make[1]: *** [Makefile:668: all] Error 2
make[1]: Leaving directory '/root/chan-sccp/src'
make: *** [Makefile:615: all-recursive] Error 1
root@AS01PBX02:~/chan-sccp# nano src/pbx_impl/ast119/ast119.c
root@AS01PBX02:~/chan-sccp# make
Making all in src
make[1]: Entering directory '/root/chan-sccp/src'
make  all-recursive
make[2]: Entering directory '/root/chan-sccp/src'
Making all in pbx_impl
make[3]: Entering directory '/root/chan-sccp/src/pbx_impl'
make[3]: Nothing to be done for 'all'.
make[3]: Leaving directory '/root/chan-sccp/src/pbx_impl'
Making all in pbx_impl/ast
make[3]: Entering directory '/root/chan-sccp/src/pbx_impl/ast'
make[3]: Nothing to be done for 'all'.
make[3]: Leaving directory '/root/chan-sccp/src/pbx_impl/ast'
Making all in pbx_impl/ast119/
make[3]: Entering directory '/root/chan-sccp/src/pbx_impl/ast119'
  CC       libast119_la-ast119.lo
ast119.c: In function ‘sccp_astwrap_get_channel_macroexten’:
ast119.c:3046:24: error: implicit declaration of function ‘ast_channel_macroexten’; did you mean ‘pbx_channel_macroexten’? [-Werror=implicit-function-declaration]
 3046 |                 return ast_channel_##_field(channel->owner);                                                    \
      |                        ^~~~~~~~~~~~
ast119.c:3068:5: note: in expansion of macro ‘DECLARE_PBX_CHANNEL_STRGET’
 3068 |     DECLARE_PBX_CHANNEL_STRGET(macroexten)
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated due to -Wfatal-errors.
cc1: some warnings being treated as errors
make[3]: *** [Makefile:581: libast119_la-ast119.lo] Error 1
make[3]: Leaving directory '/root/chan-sccp/src/pbx_impl/ast119'
make[2]: *** [Makefile:1109: all-recursive] Error 1
make[2]: Leaving directory '/root/chan-sccp/src'
make[1]: *** [Makefile:668: all] Error 2
make[1]: Leaving directory '/root/chan-sccp/src'
make: *** [Makefile:615: all-recursive] Error 1
`

2br-2b, 2025-05-09:
When installing asterisk, did you `sudo make install-headers`? I'm looking thru my list of commands and I think that's what I had to do to get it working

2br-2b, 2025-05-09:
@thijsvdc I think your issue is more closely related to #609 than this issue